christianhaitian/arkos #1062: RGB20SX no left Channel Audio - Possible Hardware issue, but with a software / drive fix?

Chillimonster68, 2024-05-08:
No left channel audio on RGB20SX.

Suspect this is a hardware issue with the internal speaker setup as it works fine through HDMI, but not with wired headphones / internal speaker.

Tested and confirmed on ARKOS and JELOS .

Can be reproduced with 240PEE test suite on GB / GBA etc

Can get round it by using DSP plugin mono in retroarch.

Could also be 'fixed' with the option to run the OS sound drivers in mono mode (Maybe a switch / tool to switch between stereo output and mono output in options / advanced etc)

christianhaitian, 2024-05-08:
I'm not supporting the rgb20sx unit.  You can try replacing the `/home/ark/.asoundrc` file with the one [here](https://github.com/christianhaitian/rk3566_core_builds/blob/master/shell-scripts/353-only/.asoundrc) and see if that resolves your issue.  Other than that, I have no other suggestions.

closed 2024-05-08 by christianhaitian

jjimenez22991, 2024-06-23:
This https://github.com/christianhaitian/arkos/issues/1062#issuecomment-2101260708 fix the issue with the sound missing channel thanks @christianhaitian