christianhaitian/arkos #1063: BUG: Enabling second SD card removes Tools from Options menu selection

riedyw, 2024-05-08:
When switching ROMS to second SD card the Tools selection is no longer visible when in Options

My second SD card did not have any files within the /tools folder so I copied the files from SD1:/roms/tools to SD1:/tools 
I then restarted the RG353V and Tools was not found under Options

I've seen this behavior in both 01272024 and 04242024. The device was RG353V

christianhaitian, 2024-05-08:
Please post a picture of the contents of your sd2 tools folder

christianhaitian, 2024-05-09:
I can't replicate this issue on my end.

christianhaitian, 2024-05-10:
Any update on on what I requested here in order to assist in troubleshooting the issue?https://github.com/christianhaitian/arkos/issues/1063#issue-2286306279

mdeguzis, 2024-05-20:
I got this as well today [image]

mdeguzis, 2024-05-20:
I was able to run my emmc flash via file manager to get myself going.

christianhaitian, 2024-05-20:
Did you try a reboot after switching to the 2nd sd card?

mdeguzis, 2024-05-20:
> Did you try a reboot after switching to the 2nd sd card?

That worked. Didn't seem to before, but good now :+1:

closed 2024-05-25 by christianhaitian