christianhaitian/arkos #1064: BUG Controllers Configuration

czapa86, 2024-05-11:
Hello,

i recently opened issue when i cannot get xbox wireless controler to work properly on ArkOS.
I was able to solve the issue. There are two main reasons that it doesn't work. 
1. Menu -> Configure Input doesn't save:
input_l_btn = "x"
input_r_btn = "y"
for ALL controllers. These two parametrs are not included in the config file for retroarch in /udev/ folder for all controllers, even for the device itself (i additionally tested dualshock 4 & xbox series s controllers). It is causing that select+start+L1+L2 button combination don't bring retroarch menu.
2.  Menu -> Configure Input saves xbox series s controller as:
input_vendor_id = "0"
input_product_id = "0"
where it should recognize it as
input_vendor_id = "1118"
input_product_id = "2835"
There was also problem with recognizing L2 and R2 buttons but only on Xbox controllers. 
I'm attaching correct config file for xbox series contollers to put in /udev/ folder
[Xbox Series X Controller.zip](https://github.com/christianhaitian/arkos/files/15284529/Xbox.Series.X.Controller.zip)

christianhaitian, 2024-05-11:
Oh great.  Thank you much.

christianhaitian, 2024-05-25:
This has been included with today's update.

closed 2024-05-25 by christianhaitian

stealthrabbi, 2024-10-22:
> This has been included with today's update.

I am experiencing this issue on a RGB30 with the latest ArkOS and an Xbox One controller. Was this update applied to the build? Not sure what I can do since mapping in retroarch doesn't save the file.