christianhaitian/arkos #1067: The hotkey FN using for fast forward in Retroarch by default

VolodymyrDA, 2024-05-14:
The hotkey FN (button 16) using for fast forward in Retroarch by default.
It conflict with retroarch menu button
Please remove it.

christianhaitian, 2024-05-14:
What unit are you referring too?  I don't know of any supported devices where a Function key is set up as a particular hotkey in Retroarch by default.

VolodymyrDA, 2024-05-14:
r36s

christianhaitian, 2024-05-14:
I don't support the R36S.  You'll need to go here for support for the R36s using ArkOS. https://github.com/AeolusUX/ArkOS-R3XS

closed 2024-05-14 by christianhaitian

VolodymyrDA, 2024-05-14:
Aeolus says that its from you image
https://github.com/AeolusUX/ArkOS-R3XS/issues/25

christianhaitian, 2024-05-14:
That's not what he's saying at all.  He's saying he will not make that change globally in his fork.  You will have to make that change in Retroarch directly for your use case.  Just go to the Retroarch settings menu>Input>Hotkeys>Fast-Forward (Toggle or Hold) then press Y.  Then go back to the root retroarch menu and go to configuration file then do Save Current Configuration.

DjSkeejay, 2025-02-19:
Thank you, I do like the fast forward hot-key once in awhile in certain games, What is a good button combo for fast forward to change it too?