christianhaitian/arkos #1068: Add Chinese Games search with  first letter of pinyin

ChestnutHeng, 2024-05-15:
Added Chinese search support for Chinese players. For example, "宝可梦" can be searched through "BKM"，the first letter of each Chinese pinyin in the word.

为中国玩家添加中文搜索支持。比如“宝可梦” 可以通过“BKM”搜索。即单词中每个汉字拼音的首字母

ChestnutHeng, 2024-05-15:
Or is there an API interface that can specify a search alias for the ROMs, and I can generate an index file for the ROMs?

christianhaitian, 2024-05-15:
I'm not sure I fully understand the question but if you're looking to search all your games, just press start>game collection settings>automatic game collections>all games.  After emulationstation refreshes the screen, you can then press the y button to search your games.

christianhaitian, 2024-05-25:
Any further update on this request?

ChestnutHeng, 2024-05-28:
For example, There is .xml files in my TF2 roms:

<?xml version="1.0"?>
<gameList xmlns:xsi="make by G.R.H">
	<game>
		<path>./Armor..Attack (World).zip</path>
		<name>Z 装甲奇袭[ZJQX]</name>
		<image>./image/Armor..Attack (World).png</image>
		<video>./video/Armor..Attack (World).mp4</video>
	</game>
.....

I want to search "ZJQX" to find this game。 I can't search that in "all games"

ChestnutHeng, 2024-05-29:
xml file：“BKMHBS”
[image]
game list：“BKMHBS”
[image]
search：“BKM”
[image]
not found：
[image]

ChestnutHeng, 2024-05-30:
@christianhaitian updated

christianhaitian, 2024-06-15:
Sorry, I don't know how to resolve this issue.

closed 2024-06-15 by christianhaitian

yjlcoder, 2024-08-16:
Hey @christianhaitian Sorry for responding to a closed issue but I think I can give an explanation and possibly create a PR for this feature.

Currently, game searching doesn't work well with non-Latin languages, especially CJK (Chinese, Japan and Korean) character set. With an English keyboard it's impossible to search rom names in CJK. A common solution is to latinize CJK character set so there's a way to index the names and make them searchable.

I'll use Chinese characters as the example 'cause I'm mostly familiar with it. "Pinyin" is a mapping that latinize Chinese to make them usable in modern keyboard and pronounceable by those who don't know Chinese language. Most of the daily Chinese names you see is Pinyin.

Since Pinyin tends to prolong the word length, typing fullnames could be very tiring. The popular way is to extract the first character of Pinyin result for each character and use that as a searching keyword. In OP's example, `宝可梦 (Pinyin: Bao Ke Meng; English: Pokemon)` should show in the result if you search `BKM`, and `Z 装甲奇袭 (Pinyin: Z Zhuang Jia Qi Xi)` should be searchable by `ZZJQX`.

To achieve we need to compile a pinyin database and index the latinized names. There are datasets on Github ([Example1](https://github.com/zhuyie/pinyin-db/blob/main/data/HanyuPinyin.txt) [Example2](https://github.com/xiaohk/pinyin_data/blob/master/pinyin/pinyin.json)) we can utilize the make the database.

Let me know if you wanna this feature. I can pick it up and give it a try.

christianhaitian, 2024-08-21:
Hi, sure if it's something that can be incorporated into emulationstation via PR, i'd be happy to review it, test it to confirm nothing else breaks as best as I can, merge it and deliver it in a future update.   The source for emulationstation used in ArkOS is here: <https://github.com/christianhaitian/EmulationStation-fcamod/tree/503noTTS>