christianhaitian/arkos #1071: ArkOS won't format 2nd SDcard correctly

s1nistr4, 2024-05-17:
If you install arkos, then go to Options > Advanced > Switch to 2nd SDCard it formats the sd cards, fails to do it correctly and creates an inaccessible, generic Linux partition on the 2nd sd card that can't be mounted even in Linux, my guess is that something went wrong formatting the sdcard and it corrupted the partition. With this issue you can't use a 2nd sd card to install ROMs. There's no info in the wiki about this and also no warning that it will wipe all your sdcard information had something been on it. This issue makes ArkOS unusable on RG353V as of 5-17-2024.

christianhaitian, 2024-05-17:
Per the FAQ here:https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353v#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games
ArkOS doesn't format the 2nd sd card at all.  It accepts EXT4, ntfs, fat32, and exfat pre-formatted sd cards.  The sd card was in a bad state when you inserted it in the first place.

closed 2024-05-17 by christianhaitian

s1nistr4, 2024-05-18:
> Per the FAQ here:https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353v#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games ArkOS doesn't format the 2nd sd card [...]

It was a generic ext4 partition and works fine on other OSes, it is only corrupted after doing precisely what the faq says.

christianhaitian, 2024-05-18:
Were you using the sd card with JELOS or ROCKNIX?  If so, there's an issue with using the 2nd sd card that was structured for use with either one of those OS that will be resolved with a OTA update later this month.