christianhaitian/arkos #1074: changing boot logo on emmc

jadedplants, 2024-05-21:
(RG353V, v2.0 04/24/2024)

I'd like to change the boot logo on ArkOS booting from emmc. I have done it successfully on ArkOS booting from TF1, but I'm not sure how to replace the image file from emmc. I have tried to replace the file via File Manager, moving the file from SD card on TF2 and pasting it into BOOT on TF1, but I think maybe there is a permissions issue? Thank you kindly!

christianhaitian, 2024-05-21:
You'll have to place the IMAGE file in the emmc BOOT partition using the sudo command.  Best solution to accomplish this will be to do this via a SSH session (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353v#q-how-do-i-ssh-into-ArkOS) or via command line with a keyboard directly connected to the unit (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353V#q-how-can-i-access-a-terminal-physically-on-arkos).

RumblePen, 2025-12-25:
Since this is the top google result I would like to add an easier way to do this for those unfamiliar with shell/ssh/terminal commands. You can do this via the ArkOS file browser. Enter your device's IP into your web browser. Select Settings>User Management. Click the Pencil icon to edit User "ark". Change Scope from "/roms2" to "/". 

Now you can drag and drop Image to /boot.