christianhaitian/arkos #1076: [feature request] Hide extraneous folders/files for Dreamcast game list

mdeguzis, 2024-05-21:
1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

No

5. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### Describe the Issue (If applicable)

Similar to PS1 games (M3U), when you add the dreamcast game folder, you get extra folders/files to browe in the Dreamcast list. Currently, I saw we cannot use .7z archives. It's likely due to the es_systems spec of :

```
<extension>.7z .7Z .gdi .GDI .cdi .CDI .cue .CUE .chd .CHD</extension>
```

I have tracks ending in .bin, a .cue, and a .gdi.

### How can the issue be reproduced? (If applicable)

Add a dreamcast game folder with tracks, bin, cue, gdi. 

1. Go to Dreamcast
2. Click on game folder
3. 2 files are present

### What version of ArkOS are you on?
04242024

### Anything else you'd like to include that may help to review this issue or feature request?

mdeguzis, 2024-05-28:
Thank you so much

christianhaitian, 2024-06-28:
This has been accommodated with today's update.

While in a system, press the start button then go to UI Settings then go to Theme Configuration then File Extensions

closed 2024-06-28 by christianhaitian

mdeguzis, 2024-06-28:
It seems to still show the folder above the files. Expected?

christianhaitian, 2024-06-28:
Yes, that's expected.