christianhaitian/arkos #1078: Can't access Mupen64(plus or reg) settings on RGB30

Dochartaigh, 2024-05-23:
### Describe the Issue (If applicable)
On the newest ArkOS, on a Powkiddy RGB30, there is NO way (I've found) to access the settings in-game. The normal Select+X does not work (nor any other hotkey combo I've used for the last 10 years).

### How can the issue be reproduced? (If applicable)
On your RGB30 open any N64 game, on the Mupen64 emulator (retroarch/retroarch32/plus or non-plus), and try to access an in-game menu.

### What device are you using? 
Powkiddy RGB30, V2 I believe (has the newer colored buttons, just bought this week off Amazon)

### What version of ArkOS are you on?
Newest "RGB30 | (Updated 10/16/2023) MD5:0C65C5C7D7CFEE353AE9FB30B42B2159"

### Anything else you'd like to include that may help to review this issue or feature request?
So sorry for such a simple question... but just spent the last hour googling this and I can't for the life of me find out how to access the settings for Mupen64 or Mupen64Plus in-game. Tried every hotkey combo I can think of... just want to try to get better performance by changing video settings (frameskip or whatnot, scaling, etc), adding a shader, show the FPS, etc. etc.

christianhaitian, 2024-05-23:
Are you sure you're not using the standalone emulator which does not have a menu?  That's the default n64 emulator in ArkOS.

Dochartaigh, 2024-05-23:
> Are you sure you're not using the standalone emulator which does not have a menu? That's the default n64 emulator in ArkOS.

Thanks for replying - that could very well be the case. I normally enter that in-game menu and click on info to see what exact flavor/version it was running so I can better google a solution myself – and since I couldn't do that I googled and it said it was Mupen64. Just watched a video and saw the Start > Emulator Settings menu where I saw it's "standalone-glide64mk2" exactly like you said!

So I know N64 is buggy (even on many higher power devices than my RGB30!)... but people are ALWAYS talking about trying different emulators, and tweaking all these different settings in order to get a select game to play better. AND they also say that "standalone-glide64mk2" is the best/fastest to use on ArkOS... so what am I missing here? Did I just read all those posts wrong? (which could very well be the case, not all were about the RGB30 in particular). 

How are people tweaking these settings if there's no menu to do so? Or are they manually editing config files? (all I'm wanting on n64 so far is an 8:7 screen ratio to better fit the RGB30's square screen + a simple shader like ZFast-CRT.

If this isn't possible, no biggie. I only play a select few N64 games anyway (and have a real console N64 if I ever get the jones for a long play session ;)  ...or I can see if people are mistaken in performance on some of the RetroArch N64 cores where I can find these options - and I'll then use those. Thanks again!

closed 2024-05-23 by Dochartaigh