christianhaitian/arkos #1079: With Arkos in 04242024, device R36S,  Saturn yabasanshiro :  not find libglut.so.3 and workaround : apt-get install freeglut3

Martossien, 2024-05-23:
The "standalone-bios' is yabasanshiro emulator 
Dont launch at the GUI
if i try in ssh in root in wifi :
/opt/yabasanshiro/yabasanshiro --?
result :
/opt/yabasanshiro/yabasanshiro: error while loading shared libraries: libglut.so.3

I install it :
apt-get install freeglut3

Now it s works !

It s work in other last version, i don t know when it don t work .

christianhaitian, thank s a lot for all !!! if you come in the north of France i promise a beer or other drink you like !!!

christianhaitian, 2024-05-23:
Hi, I don't directly support that unit so if you're using the AeolusUX version, report the issue here so he can resolve it for a future update.  https://github.com/AeolusUX/ArkOS-R3XS

Thank you for the compliment.

closed 2024-05-23 by christianhaitian

AeolusUX, 2024-05-25:
> Hi, I don't directly support that unit so if you're using the AeolusUX version, report the issue here so he can resolve it for a future update. https://github.com/AeolusUX/ArkOS-R3XS
> 
> Thank you for the compliment.

Thanks for this I would add this to the new update. Cheers!