christianhaitian/arkos #1080: Scraping "only missing medias" scrapes EVERYTHING (instead of 'only missing')

Dochartaigh, 2024-05-24:
### Describe the Issue (If applicable)
Scraping with the "only missing medias" option scrapes the ENTIRE game directory, and not just games with missing media.

### How can the issue be reproduced? (If applicable)
Start > Scraper > Scrape Now (have my ScrenScraper.fr user/pass setup already) > Filter: "only missing medias" > Select your system(s) --only 1x in my case > Start... and you'll see it's re-scraping EVERYTHING all over again and not just the SINGLE game I added. 

### What device are you using? 
Powkiddy RGB30 V2 (assume it's v2 since I have the newer buttons)

### What version of ArkOS are you on?
Newest "RGB30 | (Updated 10/16/2023) MD5:0C65C5C7D7CFEE353AE9FB30B42B2159"

### Anything else you'd like to include that may help to review this issue or feature request?
Only making this ticket because this can literally add DAYS of extra scraping... Last night I added 1 single game to my NES rom folder... should have taken ~30 seconds to download the box art and video for that ONE single game... but instead took me ~8 HOURS to scrape my entire NES folder last night (~8K games... even have 7x threads on ScreenScraper.fr...).

Also to note: After this happened, I imaged a new version of ArkOS onto another SD card and this same issue persists. The ONLY things I did on the new setup was turn on WiFi, then tried this.

christianhaitian, 2024-05-24:
Also, make sure you do an OTA update to the latest version of ArkOS which 04/24/2024 as of the date and time of this comment.

Dochartaigh, 2024-05-24:
> Also, make sure you do an OTA update to the latest version of ArkOS which 04/24/2024 as of the date and time of this comment.

Thanks again for replying! (and just want to tell you how refreshing it is to have a developer so active in troubleshooting - you don't know how amazing that is from the end-users viewpoint! AND super-rare too!!! THANK YOU THANK YOU).

Back to the question: I have to admit I'm not 100% sure... but these might (probably?) were previously scraped with Rocknix. I was told you can 'seamlessly' swap the system/smaller 16GB card from OS to OS and leave the larger 2nd game SD card in the device (which keeps gamelists and all the scraped media) and it'll 'just work' between any OS on these dual-SD-card devices, and you do NOT have to re-scrape or regenerate rom lists or anything... but sounds like that may not be the case?

I DID notice "gamelist.xml.old" being in each individual rom folder when I switched from Rocknix to ArkOS (I'm a convert to yours - honestly just seems better... and I can do SIMPLE things like change the es_systems.cfg on yours... which is 1000% locked-down on Rocknix and impossible to change, for example) ... and it might have been Arkos replacing/archiving the gamelists into that .old format, but I'm not positive. The old vs. the new gamelist files are the same exact size down to the byte fyi... so think they're still the same exact files.

Ayway, I pulled up my NES gamelist, and besides NOT having all the descriptors yours has (developer, game description, number of player, etc.), mine DO seem to be in the same exact format as yours otherwise with: "Path, Name, MD5", with "</game><game>" separating them per usual:

`<?xml version="1.0"?>
<gameList>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Duck Hunt LCD Mod (Hack) Beta 5 neslcdmod.com.nes</path>
		<name>Duck Hunt LCD Mod (Hack) Beta 5 neslcdmod.com</name>
		<md5>583878572f76623e9b6297a0fb373f68</md5>
	</game>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Hunt (Unl).nes</path>
		<name>Hunt (Unl)</name>
		<md5>ce708bd0b09e62fb78c816e131162915</md5>
	</game>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Invader (Unl).nes</path>
		<name>Invader (Unl)</name>
		<md5>7c9c63112033f7949029eebf530d63c8</md5>
	</game>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Shoot UFO (Unl).nes</path>
		<name>Shoot UFO (Unl)</name>
		<md5>10b158580759dfd09eeafedbe3fc7d33</md5>
	</game>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Snowfield Shoot (Unl).nes</path>
		<name>Snowfield Shoot (Unl)</name>
		<md5>dc9cb04db66e6cbd592fb82f9bfbc2fa</md5>
	</game>
	<game>
		<path>./---Game Series Collections/Accesories &amp; Add-Ons/Light Gun Zapper/Duck Hunt Based Unlicensed/Space 2050 (Unl).nes</path>
		<name>Space 2050 (Unl)</name>
		<md5>e4201f659d56cc4a06919d833bda89f9</md5>
	</game>`

I also ran a little test. Since when I tried to scrape that ONE single rom last time, and it re-scraped the entire 7,000+ game library, I thought it's safe to assume that ALL the scraped media and gamelist are now in the proper/official ArkOS format, right? So to test this I added another SINGLE game to NES, and re-scraped with "only missing medias" – and it again started scraping the ENTIRE system, and not just that one game!

I'm now going to re-flash my 3rd image of ArkOS onto another fresh 16GB card. After it boots up for the very first time I'll run the 'Options > Update' as the VERY first thing (that's the "OTA update" you mentioned, right? I don't know that "OTA" stands for). Before inserting my SanDisk 256gb 2nd game SD card, I'm going to take a smaller rom folder and delete all the media/art/gamelist files so it's fresh/untouched. I'll then scrape regular (EVERYTHING), reboot, then add a single game, and re-scraped with "only missing medias" to see if it only does that ONE game, or mistakenly does everything (like it's done the last several times). Does that sound like a good test? I can film it if you want... have a feeling it's going to do the same thing as the last 2x ArkOS images I flashed (one was fully updated, the last wasn't). But I'm sorry to say, this being my 3rd time imaging the same ArkOS I have little hope it'll magically start working a different way... (sorry, don't want to be negative... just factual/logical).

christianhaitian, 2024-05-24:
Oh and OTA update is short for Over The Air update.

christianhaitian, 2024-05-27:
Any update on this?

Dochartaigh, 2024-05-29:
> Any update on this?

Sorry, was down the shore for the holiday weekend. And YES - you are right, it's something in the Rocknix-created gamelist.xml (I think) which is screwing it up. I added a whole other system I had never scraped, then scraped it, then added 1x single game and re-scraped and it DID scrape JUST that game. I also did a fresh SD card flash of your ArkOS too... but don't think that was it, it was that gamelist (despite the code looking nearly exact to each other at first glance...).

closed 2024-05-29 by Dochartaigh