christianhaitian/arkos #1084: Can not flash RG353PS - always get blank screen.

MarcelvandenElshout, 2024-05-26:
Hi,
I have tried on several occassiosn to flash ArkOS onto my Anbernic RG353PS...but i can not get it to work no matter what i try!
So far i have tried:
-Image "RG353M - [GDrive](https://drive.google.com/file/d/1EZlBLjHz9OCxaflck2hzIOEdYR-wcFZy/view?usp=sharing) | [Mega](https://mega.nz/file/vUxjmRDC#V21ZrCeHqp-qAjvL4j3YdGBex7g12T3-rVR0L0WVbFc) | (Updated 01/27/2024) MD5:14BC64AA09885AFB143F85725B28CF56"
-Using a freshly formatted A-class card (Samsung Evo 32GB)
-Tried flashing with RUFUS and with Balena Etcher
-Tried the rename trick as mentioned here: [click](https://www.reddit.com/r/SBCGaming/comments/14argif/ghost_image_in_arkos_rg353p/)
-Turned on he device with SD card inserted (no other cards). Waited 30 min. Nothing. (tried several times).

Can anyone please help me with this! :) Is the latest RG353M image incompatible with RG353PS perhaps?

christianhaitian, 2024-05-26:
The rg353ps is not supported.  You're welcome to try a workaround here https://github.com/christianhaitian/arkos/issues/842

closed 2024-05-26 by christianhaitian

MarcelvandenElshout, 2024-05-27:
> The rg353ps is not supported. You're welcome to try a workaround here #842

Hi, thanks for the reply.
This is really frustrating for me, since if i read all sorts of guides and replies on Reddit literally NO ONE states that you need this file in order for it to work. I will try this suggestion ASAP.

Question though:
By default the RK3566 (according to specs) runs on 1.8 GHz with Anbernic's stock firmware. Is the 1.4 GHz speed only changed at boot? Are there files for 1.2 or 1.5 GHz as well?

christianhaitian, 2024-06-03:
See here for another solution you can try. https://github.com/christianhaitian/arkos/issues/831#issuecomment-2145811688