christianhaitian/arkos #1085: PPSSPP rum games error

baruchlinton, 2024-05-26:
![IMG_20240526_053254_800]
After patch 29032024 psp stopped working in rk2023 with arkos, the games do not start and in retroarch it asks for the "assets, folder flash0, lang, and the files ppge.atlas.zim and ppge.atlas.zim(md5)" in the bios folder but it does not recognize them being there.

christianhaitian, 2024-05-26:
I can't replicate this issue on my end.  PSP games start just fine on my end with the recent 05242024 update with both the standalonw and retroarch emulator. Try deleting the ppsspp subfolder so ArkOS can recreate it and see if that resolves your issue.  Be aware that this will delete saves and savestates for psp if you do this.

baruchlinton, 2024-05-27:
solve the problem, apparently my device had problems updating and the data in the path "opt/ppsspp and opt/scummvm" were corrupt, I never noticed the latter as it did not have games in its folder, I reinstalled the OS and the problem was solved. It would be a good idea to have a tool to check installation data after updates unless one already exists. Thanks for the time to review my problem.

closed 2024-05-27 by christianhaitian