christianhaitian/arkos #1086: Ports not work.

fabinhojcr, 2024-05-26:
Ports stopped working on r36s after the last update, I participate in an Arkos group for r36s on Telegram and many people reported the same problem there after this latest update.

elmoiv, 2024-05-27:
Happened for me too

fabinhojcr, 2024-05-27:
Broke Thememaster too.

elmoiv, 2024-05-27:
I am gonna flash older ArkOS image till they fix the problem

christianhaitian, 2024-05-27:
I don;t support the R36s units.  You can report the issue here: https://github.com/AeolusUX/ArkOS-R3XS
Also note, there's a tool (Change Ports SDL) included in the most recent update in Options/Advanced section that can be used to change the SDL back to an older version for Ports.  Thememaster wasn;t impacted with the latest update.  Thememaster was impacted by an update from today by the maintainer.  He is aware of the issue and will address it when he can.

closed 2024-05-27 by christianhaitian