christianhaitian/arkos #1090: [Feature Request] Rumble support for standalone mupen64plus

upwhatever, 2024-06-01:
Would it be possible to add rumble support for rk3326 devices to standalone mupen64plus? Looking at the mupen code, I assume one would have to patch the rumble routines and ControllerCommand in [mupen64plus-input-sdl file src/plugin.c](https://github.com/mupen64plus/mupen64plus-input-sdl/blob/master/src/plugin.c).

Btw: How is rumble on these devices triggered nowadays? Looking at https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/patches/mgba-patch-rumble.patch made me think you are writing directly to `/sys/class/pwm/pwmchip0/pwm0/duty_cycle`, but I cannot export pwm0 to sysfs on my R36s (resource busy error) and rumble with the provided retroarch rumble cores is working fine anyway.

christianhaitian, 2024-06-01:
For R36s, it's a clone of the rg351mp where rumble is natively supported there through the kernel: https://github.com/christianhaitian/linux/commit/d69da5120066ac7e65dcc4549ba9d5ade0a1fb52

Those rumble patches are for other rk3326 devices where rumble is not natively enabled like the rg351v.  standalone emulators with rumble should natively work if it's supported.  If it doesn't I don't know why.

closed 2024-06-01 by christianhaitian

upwhatever, 2024-06-01:
My bad, I was on the wrong track. The whole issue ended up being that buttons configured for "Mempack switch" and "Rumblepack switch" did not behave as expected. Thanks to your kernel support, requested feature is already implemented and working fine.

For anyone having the same issue: Set [Input-SDL-Control1]plugin=5 in mupen64plus.cfg and rumble will work.