christianhaitian/arkos #1093: CPU governor tweaks

chronoreverse, 2024-06-02:
### Describe the Issue (If applicable)
Ondemand's defaults are too aggressively power saving.  If /sys/devices/system/cpu/cpu*/cpufreq/ondemand/up_threshold is set lower from default 95 to 85 and if /sys/devices/system/cpu/cpu*/cpufreq/ondemand/sampling_down_factor is set higher from default 1 to 150, it would ramp up and stay ramped for better performance without needing the performance governor.

However, it would be even better if Schedutil were available for selection as it can even more intelligently ramp up the CPU to the load.

### How can the issue be reproduced? (If applicable)
If using bilinear-sharp shader, rewind and preemptive frames in SNES it's immediately apparent that it's chugging if using Ondemand.

Adjusting the controls via SSH will smooth it out.

Switching the governor to Schedutil also will smooth it out.

### What device are you using?
RGB30

### What version of ArkOS are you on?
05242024

christianhaitian, 2024-06-02:
Thank you for this info.  schedutil was available at one point but had even worse performance when used so I disabled it in ther kernel.

chronoreverse, 2024-06-02:
From reading up on the old Jelos posts, at some point Schedutil was problematic but seems okay at least with the current version of Arkos.  I still expect Performance governor to be better for the more difficult emulation scenarios.

foxflyy, 2024-06-03:
Just a FYI, I was keen on tweaking cpu/gpu/dram freq to save power and keep decent performance. Soon I discovered RGB30 screen was consuming hell of power. The max brightness and min delta power is up to 800mw. The demanding game run under 1.4G and 1.8G delta is only 20-30mw. So I do not care the freq anymore lol.

christianhaitian, 2024-06-03:
Yes that screen while beautiful is extremely power hungry.  Because of the scrreen, it has the worst battery life of any rk3566 handheld unit so far.

foxflyy, 2024-06-03:
> Yes that screen while beautiful is extremely power hungry. Because of the scrreen, it has the worst battery life of any rk3566 handheld unit so far.

I was learnt that this screen design is for a smart home switch mount on the wall which has very ideal brightness, but do not care about the power at all.

christianhaitian, 2024-06-03:
I appreciate what you found to improve performance on the ondemand cpu governor for the rk3566 chipset.  For the highend scenarios as you described, simply going performance does make the best sense.  Since users have the ability to set the performance on a OS wide, system wide and specific game basis, end users can choose what works best for their usage.  I will admit that I have not seen much of a battery usage difference between powersave and performance for even light gaming like gameboy.

chronoreverse, 2024-06-03:
Yeah, my motivation was noticing the back of my RGB30 got warm on the back playing GBC rather than battery life.  I don't believe for the RGB30 at least that the battery life is significantly affected.

closed 2024-06-28 by christianhaitian