christianhaitian/arkos #1099: [Feature Request] Zip support for N64 roms

upwhatever, 2024-06-07:
### Describe the Issue (If applicable)
Different from most other cartridge based systems, N64 currently does not support zipped rom files. This is most likely a result of standalone mupen64plus lacking support for zip file format. Retroarch N64 cores can already zip files.

### How can the issue be reproduced? (If applicable)
Drop zipped N64 roms into /roms/n64/. Per default they are not detected at all by ES. After modifying N64 extensions in es_systems.cfg to include zip, these roms only work with retroarch cores.

### Anything else you'd like to include that may help to review this issue or feature request?
I suggest implementing zip support for standalone mupen64plus by modifying /usr/local/bin/n64.sh launch script. One could, for example, temporarily extract such zips to a tmpfs folder before launching them with mupen64plus. Tmpfs usually resides in RAM, so this process is very quick and does not cause any additional writes to microSD card.

Proof of concept:
```
--- /usr/local/bin/n64.sh.bak
+++ /usr/local/bin/n64.sh
@@ -14,6 +14,15 @@
   game="${2}"
 fi
 
+tmpfsdir="$(dirname "$game")/tmpfs"
+if [[ $1 == standalone-* ]] && [[ "${game,,}"  =~ \.zip$ ]]; then
+  mkdir -p "$tmpfsdir"
+  mountpoint -q "$tmpfsdir" && sudo umount "$tmpfsdir"
+  sudo mount tmpfs "$tmpfsdir" -t tmpfs -o "size=75m,uid=$(id -u),gid=$(id -g)"
+  unzip "$game" -d "$tmpfsdir"
+  game="$tmpfsdir/$(zipinfo -1 "$game" | head -n1)"
+fi
+
 if [[ $1 == "standalone-Rice" ]]; then
   if [[ $2 == "Widescreen_Aspect" ]]; then
     /opt/mupen64plus/mupen64plus --resolution "${xres}x${yres}" --plugindir /opt/mupen64plus --gfx mupen64plus-video-rice.so --corelib /opt/mupen64plus/libmupen64plus.so.2 --datadir /opt/mupen64plus "$game"
@@ -37,3 +46,4 @@
   /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$3"
 fi
 
+mountpoint -q "$tmpfsdir" && sudo umount "$tmpfsdir"
```
This creates a 75MB tmpfs mountpoint in the same folder where the zipped rom is located (e.g. /roms/n64/), which should be enough for any N64 rom up to 64MB. No cleanup is necessary, unmounting tmpfs will get rid of any data stored there. There may be a better location to put the tmpfs mountpoint.

christianhaitian, 2024-06-07:
Yes I have an idea to support this similar to how compression is supported for coco games in `/usr/local/bin/coco.sh` from lines 68 - 90.

christianhaitian, 2024-06-09:
zip and 7z support will be included in this months update.
https://github.com/christianhaitian/rk3566_core_builds/blob/master/shell-scripts/n64.sh

closed 2024-06-28 by christianhaitian