christianhaitian/arkos #1101: RG353P Atari ST White Screen & Battery Warning Verbal.Voice HotKeys

jerzzey, 2024-06-10:
After A Few Days Ago I Complaint About R36S And Still Nowhere To Fix Now Today I Brought RG353P And Result Was The Same, And Arkos Was Update And Still Stuck, And There Is No Core In MetaData How Can I Fix It? And What A ShortCut Hot Keys For Verbal Sound Battery Warning? I Try FN+R3 Like RG353V But No Success Already Try FN+Cross, Square, Triangle, L1/2, R1/2, Select Start And Still No Sound

christianhaitian, 2024-06-10:
The RG353P units are not supported.  For a list of supported devices, see the main page of the wiki.  For list of hotkeys, check the FAQ on the wiki for a supported device.  RG351M is no longer supported as of 2021.  Did you mean RG35**3**M?  Also, have you tried to update the retroarch cores in Retroarch?  Go to the Retroarch system, launch retroarch, then do core downloader, then update cores.

closed 2024-06-10 by christianhaitian

jerzzey, 2024-06-10:
> The RG353P units are not supported. For a list of supported devices, see the main page of the wiki. For list of hotkeys, check the FAQ on the wiki for a supported device. RG351M is no longer [...]

Yea Sorry Its RG353M Image, Btw I Found Bout The Battery Warning Its Select+R3 But Atari St Remaining White Screen, Already Up All Cores, Dun Know If Its Because All File Is .IPF Or Something Bug That Never Fix, Kinda Tired Of Searching This Atari St Problem Because My 2 Console Remain The Same, Maybe Will Brought RG552 Hope Better With Android System