christianhaitian/arkos #1105: PPSSPP Menu inaccessible & mapping issues

EchoFiend2000, 2024-06-12:
Device: RG353M
ArkOS Version: 052424

Recently updated to the newest version and can't seem to access the PPSSPP menu using the R3 hotkey. 

I found a manual fix but will leave this here incase anyone has a similar issue. 

I noticed that the controls.ini file on the EASYROMS partition of my SD card 1 was different from the controls.ini file on my Roms2 SD card 2. I just copied and pasted the 'pause' value from 1 to 2. (In this case it was 10-107.) 

I also noticed that circle has been mapped to B, X to A, Square to X and Triangle to Y, but I just changed these manually from the PPSSPP menu. I also noticed that the default menu hotkey seems to be select and l2 but this wasn't working for me either (allow combo mappings was off). 

I also updated the controls for games with game specific settings manually using the PPSSPP menu as these were unaffected by changing controls.ini.

Not sure if there is an easier fix for this like 'Fix global hotkeys' or 'Restore default PPSSPP controls' or whether I should continue to leave this open. 

Just want to say thank you so much for all the hard work that you do on ArkOS, these devices would be nothing without your software. 

Thanks

christianhaitian, 2024-06-12:
Thank you for the kind words, however, all this is not necessary because the tool in the Options>Advanced system section named `Restore Default PPSSPP Controls` should take care of that.  Based on your description of the control issues, did you have a previous ArkOS PSP install from another device that you copied the PSP folder from?

EchoFiend2000, 2024-06-12:
Yes that must be exactly the cause of the issue! Thank you. My second SD is from my previous RK2023. 

I used restore default PPSSPP controls for good measure and everything is functioning perfectly now. 

Is this solution permanent or is there something else I can do to prevent the controls being changed back to the RK2023 configuration the next time I update?

Thanks again

christianhaitian, 2024-06-12:
It won't change back automatically so you should be good moving forward.  Enjoy.

closed 2024-06-12 by christianhaitian