christianhaitian/arkos #1109: [Question]Palm game for noob

Fr-Dae, 2024-06-17:
Good morning,
I've probably done something wrong, but I can't run the downloaded games for palm os.

I have checked that I have the latest core for palmOS(mu).
Nothing is displayed in the main menu with the other systems.

I also tried in "retroarch" mode by manually selecting the zip of the game and the core, nothing launches.

If you have any ideas, I'm interested.

christianhaitian, 2024-06-17:
Did you provide the bios file necessary for Palm per the information here: https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#palm-os?

Are you making sure the games you're trying are not ones that need Palm OS 5.x?  Palm OS 5.x is not yet supported by the emulator.

Fr-Dae, 2024-06-17:
Hello, after rebooting, the PalmOS menu appears, but the games don't work.

I have to check with retroarch if I have palmos(mu) and mu_libretro. 
Both are present and up to date.

Do I need to do anything in particular? Do you have more detailed instructions?

Fr-Dae, 2024-06-17:
It is Palm OS 4( and 3 some game)
and I've just tested, they only appear if the zip is decompressed in a folder

christianhaitian, 2024-06-17:
As per the Emulator and Ports section states, only files with extensions of .img .IMG .prc .PRC are supported.  And do you have the bios file loaded in the bios folder?

Fr-Dae, 2024-06-17:
could you give me the exact path folder to check that please ?
because i know we have a bios folder, and a retroarch/core folder

christianhaitian, 2024-06-17:
Per bulletpoint 3 in the Emulator and Ports section:

All bios files go into the bios folder located in the roms (EASYROMS) folder unless specifically stated otherwise in the emulators list below. For those devices that support a dual (2) sd card setup (ex. RG351V or RG353M), if you're using a dual (2) sd card setup, all bios files go into the bios folder located on the 2nd sd card.

Fr-Dae, 2024-06-17:
```properties
ark@rg351mp:/roms2/bios$ tree ./
./
├── dc
│   ├── data
│   ├── dc_nvmem.bin
│   ├── vmu_save_A2.bin
│   ├── vmu_save_B1.bin
│   ├── vmu_save_B2.bin
│   ├── vmu_save_C1.bin
│   ├── vmu_save_C2.bin
│   ├── vmu_save_D1.bin
│   └── vmu_save_D2.bin
├── mame
│   └── hash
│       ├── advision.xml
│       ├── apple2_cass.xml
│       ├── apple2_flop_clcracked.xml
│       ├── apple2_flop_misc.xml
│       ├── apple2_flop_orig.xml
│       ├── apple2gs_flop_clcracked.xml
│       ├── apple2gs_flop_misc.xml
│       ├── apple2gs_flop_orig.xml
│       ├── astrocde.xml
│       ├── coco_cart.xml
│       ├── coco_flop.xml
│       └── megaduck.xml
├── mame2003
│   └── hiscore.dat
├── mame2003-plus
│   ├── artwork
│   └── hiscore.dat
├── mame2010
│   ├── artwork
│   ├── crosshairs
│   ├── fonts
│   └── samples
├── scummvm
│   └── theme
│       ├── gui-icons.dat
│       ├── residualvm.zip
│       ├── shaders.dat
│       └── translations.dat
└── scummvm.ini
```

```properties
ark@rg351mp:~/.config/retroarch/cores$ tree ./
./
├── 00_example_libretro.info
├── 2048_libretro.info
├── 3dengine_libretro.info
├── 4do_libretro.info
├── 81_libretro.info
├── 81_libretro.so
├── DoubleCherryGB_libretro.info
├── DoubleCherryGB_libretro.so
├── a5200_libretro.info
├── a5200_libretro.so
├── advanced_tests_libretro.info
├── anarch_libretro.info
├── applewin_libretro.info
├── applewin_libretro.so
├── ardens_libretro.info
├── arduous_libretro.info
├── arduous_libretro.so
├── atari800_libretro.info
├── atari800_libretro.so
├── bk_libretro.info
├── blastem_libretro.info
[... 338 lines cut ...]
├── vircon32_libretro.info
├── virtualjaguar_libretro.info
├── virtualjaguar_libretro.so
├── virtualxt_libretro.info
├── vitaquake2-rogue_libretro.info
├── vitaquake2-xatrix_libretro.info
├── vitaquake2-zaero_libretro.info
├── vitaquake2_libretro.info
├── vitaquake3_libretro.info
├── vitavoyager_libretro.info
├── wasm4_libretro.info
├── wasm4_libretro.so
├── x1_libretro.info
├── x1_libretro.so
├── x64sdl_libretro.info
├── xrick_libretro.info
├── yabasanshiro_libretro.info
├── yabasanshiro_libretro.so
├── yabause_libretro.info
└── yabause_libretro.so

0 directories, 379 files
```

on R36S with 2 sd card samsung EVO

Fr-Dae, 2024-06-17:
OK, could you explain exactly what I need to do to correct this?
I've checked in the retroarch core download and palmOs(mu) is up to date.

On the project git I don't understand where to look or what to download. 

Basically I need to add a file to the bios folder, is that it?

christianhaitian, 2024-06-17:
The bios file you need is listed in Emulators and Ports section that I linked to in my first response.  Search and find that file.

closed 2024-06-17 by christianhaitian

Fr-Dae, 2024-06-17:
where ?

Fr-Dae, 2024-06-19:
@christianhaitian  could you give me information please ?