christianhaitian/arkos #1110: touch screen not working (both)

Korelustahar, 2024-06-27:
Hello, I am an rg353v user. After installing the latest version of Arkos, my touch screen stopped working. I can't use it on the Android part either. Sometimes it only detects one or two touches at the first startup, then it doesn't work. I just bought the device, how can I fix it? By the way, thank you for this beautiful software you have created.

christianhaitian, 2024-06-27:
Hello,

If the touchscreen is not working in either Android or Linux then the touchscreen is physically defective.  You'll need to contact the seller your brought the unit from for a resolve.

Good luck.

closed 2024-06-27 by christianhaitian

Korelustahar, 2024-06-28:
Thank you for fast reply but i think something crushed after portmaster. I didnt droped. And before it was working.

Korelustahar, 2024-06-28:
Ok than, thank you 🙏