christianhaitian/arkos #1114: Stick Drift after updating to the latest verion ok ARK OS (R35s)

arunim1990, 2024-06-30:
There is a stick drift with the update! please fix it. It is not there however with the previous version.

arunim1990, 2024-06-30:
Issue is with version 06272024

christianhaitian, 2024-06-30:
There were no changes in the update that would have caused the stick drift.  You may need to adjust the appropriate settings in retroarch if it's a retroarch based emulator causing the issue.

closed 2024-06-30 by christianhaitian

arunim1990, 2024-06-30:
It's happening on GTA VC and GTA 3  (portMaster). Also sometimes on
Dreamcast games. Haven't tested the rest though.

christianhaitian, 2024-06-30:
The update definitely does not impact analog sticks for portmaster games.  The most likely issue is that you have a joystick that is failing.

arunim1990, 2024-07-01:
Hi i just reconfigured for ark os and it works fine. How do i configure the deadzone settings for Ports?

christianhaitian, 2024-07-01:
Global setting for deadzone adjustment is not possible at the moment.  Ports needing deadzone adjustments have to be supported on a port by port basis.  There's no global setting available for that.

arunim1990, 2024-07-01:
Ok, thanks for the clarification. If in future it is possible please update
it with the os.