christianhaitian/arkos #1115: Kid Mode with Systems Off

momenymagic, 2024-07-01:
Hi... I have been banging my head against the wall trying to get out kid mode and I thought someone might be able to help.

I enabled Kid mode and in my infinite wisdom also turned off systems. And the unit loads into favourite only. I was following a Retro Corp guide on setting up a handheld with 20-30 games with large images etc for a child.

All works great. But I can't for the life of me get out Kid Mode if I want to tweak settings etc. I know that you can use up, up, down, down, left, right, left, right, B, A to disable kid mode. But not if there is no systems visible and the handheld loads into favourite with no other menu, systems. As soon as you hit B or A a game starts. As you boot into a game list and there is no other menu structure.

Is there another trick? Button combo or holding a button when it loads? Or maybe a config file that can be edited?

Any help would be greatly appreciated!

Great OS.... Love it. Have it on 3 handhelds. Only one with Kid Mode!

christianhaitian, 2024-07-01:
To get out of kid mode, you have to do the combo using the joystick not the dpad.  B should not start a game and by the time you hit A, it should give you the full menu again.

momenymagic, 2024-07-01:
Thanks for that... I have tried that already. It definitely starts the game using joystick too. As it is in the game list menu. Using either left or right joysticks. Definitely starts the game. I have also tried changing sequence of A and B to factor for different handhelds and RetroArch configs as suggested online and still no luck. This particular HandHeld is the R36S. I understand it is a little more problematic than the Ambernics etc

There isn't a config file in ArkOS that has a setting "Kid" that can be changed to "Full"?

christianhaitian, 2024-07-02:
I don't support the R36S units, however the following should work.

1.  Hold the B button while the system is booting from a powered off state, you should then be presented with the BaRT (Boot and Recovery Tool) menu.  
2. Select 351Files, then navigate to `/home/ark/.emulationstation/es_systems.cfg`.  
3. Hit A and edit as text.  
4. Scroll down to the line that says `<string name="UIMode" value="kid" />`.  
5. Hit X then Delete line.  
6. Hit B then Save.  
7. Hit X then Quit.
8. Hit A on Emulationstation

You should have access to the full Emulationstation menu now.

closed 2024-07-02 by christianhaitian