christianhaitian/arkos #1118: GameBoy emulator does not recognize save file (.srm)

Mehuu, 2024-07-03:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki? Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Hi, so my GameBoy emulator is no longer recognizing my saved file in Pokemon Red from yesterday. Note: this is the save option from within the game itself, not the saved states feature that is built into ArkOS. I checked the rom folder for GB and found a .srm file (size ~16 KB) and it seems to be uncorrupted. However, no matter what, I am unable to get the emulator to recognize this save.

### How can the issue be reproduced? (If applicable)
Not sure, I don't remember what I did before to cause this issue.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
Officially, I am using an R36S, running the RG351MP version.

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 2.0 (05242024)

### Anything else you'd like to include that may help to review this issue or feature request?

Mehuu, 2024-07-03:
Here is a [Google Drive link](https://drive.google.com/file/d/1cv5x4nNIioYE4ImAliJseGPfcUr3UKQ4/view?usp=sharing) (I can't upload it here to GitHub) to the .srm I was able to recover. Maybe it'll be useful if someone can try to see if they are able to load it on their own emulators, and if not, it may just be an unfortunate problem with the file itself (maybe it actually did get corrupted).

christianhaitian, 2024-07-03:
Couple of items:

1.  I don't provide support for the R36s units.  You can get support from the Aeolus fork available here: https://github.com/AeolusUX/ArkOS-R3XS
2.  If a save is not being recognized, it could very well be corrupted or you may have a SD card that is about to fail especially if you're still using the original SD card that came with the unit.  I can only advise that you ensure that the retroarch cores are up to date and ensure you're using a good sd card.

closed 2024-07-03 by christianhaitian