christianhaitian/arkos #1122: French translation

chronoss09, 2024-07-04:
I use a R36S and RG10Max and i see the French translation is not finish yet, here my trad but some word can't change in the console... maybe u can fix it...
[emulationstation2 fr par Chronoss.zip](https://github.com/user-attachments/files/16093970/emulationstation2.fr.par.Chronoss.zip)

christianhaitian, 2024-07-04:
Use this for translation: https://github.com/christianhaitian/arkos/issues/1011#issuecomment-2032672947

chronoss09, 2024-07-04:
Ok thanks, i will work on it !!!

chronoss09, 2024-07-04:
Other question, in the release page, wich file i need for R36S update ?

chronoss09, 2024-07-05:
Hi !!!
Can u explan this line : **216 msgid "A,AN,THE**"

chronoss09, 2024-07-05:
ah ok, so i need to repeat samething for french langage ?

christianhaitian, 2024-07-05:
Correct

chronoss09, 2024-07-05:
ok thanks !!!

chronoss09, 2024-07-05:
Lign 544 and 560 are samething, not a mistake ?

lauren7ino, 2024-07-05:
Hi Christian, @chronoss09,

A better example maybe:
For the game "The Great Giana Sisters" on Amiga - Users search for G for 'Great', not T for 'The'. The idea is to ignore those articles when sorting.

By the way, Christian are you sure articles are translatable ?
Because that means you should also have all the game names translated, which may not be common.

Maybe French articles should be appended to the existing English ones, in case some game gets translated.

lauren7ino, 2024-07-05:
"GOVERNOR" refers to CPU Performance Governor, which I believe tries to find a balance between power saving and performance on demanding titles.

chronoss09, 2024-07-05:
> "GOVERNOR" refers to CPU Performance Governor, which I believe tries to find a balance between power saving and performance on demanding titles.

So, i mean "Puissance" is the best thing instead of "GOUVERNEUR" in French !!!

lauren7ino, 2024-07-05:
Meaning 'CPU Frequency Scaling' .... maybe you can find a translation inspired in 'Auto CPU Performance'

chronoss09, 2024-07-05:
Yeah, i think it's more better...

christianhaitian, 2024-07-05:
Only menu items are translated with this, not game names.

chronoss09, 2024-07-06:
@christianhaitian 

Here is the French trad. I have add more lines that missing in your official file, it begins at 1315 but other don't work. maybe u can do something... I put a note on ones that don't work...
[emulationstation2 fr par Chronoss.zip](https://github.com/user-attachments/files/16115130/emulationstation2.fr.par.Chronoss.zip)

chronoss09, 2024-07-07:
How to use your update file on R36S console for exemple ?

christianhaitian, 2024-07-07:
Until it comes as part of the next update, you can manually load that file to `/usr/bin/emulationstation/resources/locale/fr/` in the OS via ssh.

chronoss09, 2024-07-07:
> Until it comes as part of the next update, you can manually load that file to `/usr/bin/emulationstation/resources/locale/fr/` in the OS via ssh.

I talk about your update file on this github "arkosupdate07042024.zip", it's the last file you have released...

christianhaitian, 2024-07-07:
Oh that may not work for the R36S since I don't support the R36S units.  The updates are meant to be applied over the air.  If you're using the AeolusUX fork of ArkOS, just wait for the update there.

chronoss09, 2024-07-07:
> Oh that may not work for the R36S since I don't support the R36S units. The updates are meant to be applied over the air. If you're using the AeolusUX fork of ArkOS, just wait for the update there.

ok i see, thanks !!!

have u fixe line that not work ?

christianhaitian, 2024-07-31:
This has been accommodated with today's update.

closed 2024-07-31 by christianhaitian

chronoss09, 2024-08-01:
> This has been accommodated with today's update.

There is new line on new file ?

On last Arkos update, i see new line on setting like :
- File Extention and more...