christianhaitian/arkos #1125: ArkOS_RGB30_v2.0_05242024.img won't load on the Powkiddy RGB30.

Dunard, 2024-07-07:
Just sit there on the logo.
[image]

christianhaitian, 2024-07-07:
What is the make and model of the sd cards you’re using? If they’re are not genuine Sandisk or Samsung sd cards, try that first. If they are and your still encountering this issue, then your device just might not be fully compatible with ArkOS.

Dunard, 2024-07-08:
Trying out my orange/white Samsung 32gb sd card. I had to removed the Chrome OS installer from it.

Yeah the Micro SD HC 10 16gb that I tried is the problem. Works fine with the JelOS and another one that I won't say here.

Dunard, 2024-07-08:
Arkos screen didn't blink when using the Samsung sd card.

It did blink twice when using the Patriot sd card. Waiting to see it get pass the logo screen.

Dunard, 2024-07-08:
I gave up. I think this device can't handle it for some reason. Even it did blink twice. Stuck.

christianhaitian, 2024-07-08:
Yea that happens with some devices unfortunately.  ROCKNIX should work on your unit though.

closed 2024-07-08 by christianhaitian