christianhaitian/arkos #1126: Youtube Stream or Twitch Stream with ARKOS

djsmoke2009, 2024-07-08:
Has anyone streamed directly from the device to Youtube or Twich? I have seen the function exists but I can't start it.

Start stream simply does not appear in the menu.

thanks for further information.

christianhaitian, 2024-07-08:
What device are you using?  If it's a RK3566 device such as the RG353x or RGB30 unit, you could connect via HDMI to an external device like a capture card and stream from there.

djsmoke2009, 2024-07-08:
I have rg351v . You can enter the stream key in the menu. But there is no function to start it.

christianhaitian, 2024-07-08:
The RG351V does not support video out so you can not stream directly from the video.  There's also no youtube app for ArkOS on the device so not sure what menu you're talking about here as well.

closed 2024-07-08 by christianhaitian