christianhaitian/arkos #1129: [BUG] Random Menu Item Selection Issue on RA and ES

import-tiago, 2024-07-09:
### Describe the Issue (If applicable)
Item selection in the menus is happening on its own and randomly, both in RA and ES ([video](https://youtu.be/JXt0CVBiwcM)).

To test if it could be a hardware issue, I used it for a week with Android and, in that case, the problem did not occur.

### How can the issue be reproduced? (If applicable)
I just turn off the device, turn it back on, and start switching between themes (I'm trying to decide which one to use).

### What device are you using? 
RG353V

### What version of ArkOS are you on?
- ArkOS 2.0 (07042024)

christianhaitian, 2024-07-09:
That's a joystick deadzone issue with your device.  At least one of your joysticks is failing.  You can increase the deadzone setting in Retroarch to compensate for that but that's not possible system wide at the moment without changing some code in the kernel.

closed 2024-07-09 by christianhaitian

import-tiago, 2024-07-09:
Yes, it's a good hypothesis. However, the 'configure input' from ES seems to solve the issue, which makes me think there might be a software-related point here. But everything is working properly now. Thanks!

christianhaitian, 2024-07-09:
That shouldn't be needed but if it works for now, so be it.

import-tiago, 2024-07-11:
For anyone with similar problems:

Using the input controls test tool, I noticed a lot of output messages even without pressing any buttons, which further supports the analog drifting hypothesis. The messages were appearing too quickly to determine their trigger source, unfortunately.

To investigate further, I opened the RG353V device and unplugged both analog control flat cables. After doing this, the input controls test tool showed no output messages unless a button was pressed, confirming the issue with the analog controls.

As a final solution, I ordered Hall sensor analog controls and will be replacing them soon. [tutorial guide](https://retrohandheldguides.com/rg353v-joystick-replacement-guide/).