christianhaitian/arkos #1130: Unable to see preview images

auanasgheps, 2024-07-10:
### Describe the Issue (If applicable)
Unable to see preview images.

Hi,
I've installed the ArkOS R3XS for my R36S, which I believe has the same logic as ArkOS, hence opening the issue here.

I'm using a dual microSD setup. I've loaded ROMs and preview images on such card, but images do not appear.
The card was prepared by the device using the feature to switch ROMs on the second card.

Please note that I do not want to use scraping as I don't have a WiFi adapter.

The folder structure is
`sdcard\<systemname>\images` (but also tried `Imgs`)
Example:
[image]

The filenames correspond to the ROM names, as these both come from a curated set.

Am I missing something? #362 suggets I should use "images" subfolder for each system, as I did.

### How can the issue be reproduced? (If applicable)
Install  ArkOS R3XS
Use the option to use the second card
Load games and preview images
Games do appear, images don't

### What device are you using? R36S

### What version of ArkOS are you on?
[ArkOS R3XS V2.0 (06272024)](https://github.com/AeolusUX/ArkOS-R3XS/releases/tag/ArkOS.V2.0.06272024)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-07-10:
Make sure the theme you're using supports preview images and that it's enabled in the start menu>ui settings>theme configuration.

Then the gamelist.xml file in each system folder would need to point to each of those images.  As an example:
```
<?xml version="1.0"?>
<gameList>
	<game>
		<path>./Contra I.zip</path>
		<name>Contra</name>
		<desc>In the year 2631, a small meteorite has fallen into the Galuga archipelago, located 20km northeast off the coast of New Zealand. Two years later, a terrorist group known as Red Falcon has seized the island in preparation for an alien invasion. The earth's marines sent two members of their elite Contra unit, Bill Rizer and Lance Bean, to neutralize the terrorists.

As either, Bill (player 1) or Lance (player 2), the player must get to the end of each stage, shooting everything in sight, while avoiding enemy fire themselves. The player can upgrade their standard-issue rifle into one of four different weapons, including the powerful spread shot.</desc>
		<image>./images/Contra I-image.png</image>
		<marquee>./images/Contra I-marquee.png</marquee>
		<thumbnail>./images/Contra I-thumb.png</thumbnail>
		<rating>0.9</rating>
		<releasedate>19880202T000000</releasedate>
		<developer>Konami</developer>
		<publisher>Konami</publisher>
		<genre>Shooter / Run and Gun</genre>
		<players>2</players>
		<playcount>1</playcount>
		<lastplayed>20240214T145245</lastplayed>
	</game>
	<game>
		<path>./Contra II - Super C.zip</path>
		<name>Contra</name>
		<desc>In the year 2631, a small meteorite has fallen into the Galuga archipelago, located 20km northeast off the coast of New Zealand. Two years later, a terrorist group known as Red Falcon has seized the island in preparation for an alien invasion. The earth's marines sent two members of their elite Contra unit, Bill Rizer and Lance Bean, to neutralize the terrorists.

As either, Bill (player 1) or Lance (player 2), the player must get to the end of each stage, shooting everything in sight, while avoiding enemy fire themselves. The player can upgrade their standard-issue rifle into one of four different weapons, including the powerful spread shot.</desc>
		<image>./images/Contra II - Super C-image.png</image>
		<marquee>./images/Contra II - Super C-marquee.png</marquee>
		<thumbnail>./images/Contra II - Super C-thumb.png</thumbnail>
		<rating>0.9</rating>
		<releasedate>19880202T000000</releasedate>
		<developer>Konami</developer>
		<publisher>Konami</publisher>
		<genre>Shooter / Run and Gun</genre>
		<players>2</players>
	</game>
	<game>
		<path>./Super Mario Bros III.zip</path>
		<name>Super Mario Bros. 3</name>
		<desc>Bowser is back, and along with the Bowser kids, he's out to spoil Mario's day once again. 

This time, along with nabbing the Princess, the Bowser crew have also taken the magic wands from the Kings of each land and turned them into creatures.
It's up to Mario and Luigi to get back the magic wands, defeat Bowser, and save the Princess all before the day is done.

Old and new power-ups abound in each level for the Mario Brothers, including the return of the Super Mushroom and Fire Flower, plus the new power-ups including the frog suit, the warp flute, and the most important of all, the leaf, which allows the player to fly and attack enemies with your tail.</desc>
		<image>./images/Super Mario Bros III-image.png</image>
		<marquee>./images/Super Mario Bros III-marquee.png</marquee>
		<thumbnail>./images/Super Mario Bros III-thumb.png</thumbnail>
		<rating>0.95</rating>
		<releasedate>19900212T000000</releasedate>
		<developer>Nintendo</developer>
		<publisher>Nintendo</publisher>
		<genre>Platform</genre>
		<players>1-2</players>
		<playcount>5</playcount>
		<lastplayed>20240214T152003</lastplayed>
	</game>
</gameList>
```

auanasgheps, 2024-07-10:
I'm using ES-THEME-NES-BOX which does support images previews so it's not the issue. "Gamelist view style" is set to Automatic, changed it to Detailed but still no preview, but clearly the space's for it is there.

> Then the gamelist.xml file in each system folder

Wait... never heard of that before. How does that get generated?

auanasgheps, 2024-07-10:
Alright, sorted. 

Press Start on main menu > Advanced Settings > Search for Local Art.

I finally got my artworks, thanks for pointing me in the right direction!

closed 2024-07-10 by auanasgheps

clodi95, 2024-08-15:
> Press Start on main menu > Advanced Settings > Search for Local Art.

This does work in the game list, but it doesn't work for the loading screen (if i set the marquee as loading image, i won't see the marquee image, but the standard loading image).