christianhaitian/arkos #1131: RGB30 - SNES games don't load after migrating to Arkos (fix found, improvement sugestion)

matmoraes, 2024-07-13:
SNES games were not loading after migrating from stock JelOS to ArkOS.

This issue can be solved removing these two lines from each game in gamelist.xml inside SNES folder.

<core>snes9x</core>
<emulator>libretro</emulator>

I believe that this configs override the defaut system config, and since ArkOS doesnt recognize libretro the games are not loaded.

As a sugestion, ArkOS could ignore the gamelists core and emulator if they are not supported. Run system default instead.

christianhaitian, 2024-07-13:
Core and emulator are supported settings in the emulationstation used by ArkOS.  No good way to ignore an unsupported emulator or core without ignoring the setting. Best to just delete the gamelist.xml file in the game folder and let ES regenerate it on boot.

closed 2024-07-13 by christianhaitian

matmoraes, 2024-07-14:
If I delete the gamelist.xml it works but I lose the scraping (link to images + descriptions)

christianhaitian, 2024-07-14:
Try Start Menu>Advanced Settings>Search for Local Art