christianhaitian/arkos #1137: [Feature Request] Submenu to determine inputs order once BT controllers are connected

Lum1r, 2024-07-31:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

YES

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

N/A

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

YES

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

issue: when i am using standalone emulator (ex Rice for N64) having console plugged with HDMI and with 2 BT controllers i am unable to remap controller order in that way so internal controller is skipped and primary controllers would be bluetooth controllers. That results in fact that emulator discovers internal controller as P1 and bluetooth controllers next.

proposed solution: in different distributions (stock, jelos) based on emulationstation I have observed that "input" section in main menu has also list of players starting from P1, P2 ect. Switching binded controller in that menu in a way that P1=BT ; P2=BT P3=internal ;  results with controllers beeing properly recognised even with standalone emulators not based on retroarch.

### How can the issue be reproduced? (If applicable)

connecting bluetooth controller to the system and trying to play on it as Player1 on any standalone emulator

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RG353V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

2.0 / 07042024

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-08-01:
This may be something that's considered in the future but it's not something I'm interested in at the moment.  If someone else would like to incorporate that, it would considered for inclusion to the ES used in ArkOS.  The source for it is here: https://github.com/christianhaitian/EmulationStation-fcamod/tree/503noTTS

closed 2024-08-01 by christianhaitian

Lum1r, 2024-08-01:
thanks Christian, I appreciate You answering so quickly, hoping it some day feature described will be considered as worthy implementing. Thank You for all the effort You are placing! Cheers!