christianhaitian/arkos #1138: Cannot compile kernel and uboot

imagemlt, 2024-08-01:
Hello, when I managed to build kernel and uboot by myself I meet such error when running `make ARCH=arm64 rk3566.img`:

make[1]: *** 没有规则可制作目标“arch/arm64/boot/dts/rockchip/rk3566.dtb”。 停止。

means there is no rules for making target 'arch/arm64/boot/dts/rockchip/rk3566.dtb'. how can i solve this problem?

my historical commands:
```shell
git clone  https://github.com/christianhaitian/RG353VKernel.git  --recurse-submodules 
cd RG353VKernel/
make ARCH=arm64 rk3566_optimized_linux_defconfig -j 12 && make ARCH=arm64 KERNEL_DTS=rk3566 KERNEL_CONFIG=rk3566_optimized_linux_defconfig -j 12
make arm64 rk3566.img
```
only show errors when I run the last command.
thanks very much if recieved any of your advice.

imagemlt, 2024-08-01:
I checked arch/arm64/boot/dts/rockchip/Makefile  and didn't found rules about rk3566.dtb. My target device is the RG353P.

christianhaitian, 2024-08-01:
Try the following:

```
git clone --recursive https://github.com/christianhaitian/RG353VKernel.git
cd RG353VKernel
make ARCH=arm64 rk3566_optimized_linux_defconfig && make ARCH=arm64 KERNEL_DTS=rk3566 KERNEL_CONFIG=rk3566_optimized_linux_defconfig
```

imagemlt, 2024-08-01:
still failed again.
```

  AS      arch/arm64/kernel/vdso/note.o
  AS      arch/arm64/kernel/vdso/sigreturn.o
make[1]: *** 没有规则可制作目标“arch/arm64/boot/dts/rockchip/rk3566.dtb”。 停止。
make: *** [arch/arm64/Makefile:145：rockchip/rk3566.dtb] 错误 2
make: *** 正在等待未完成的任务....
  LD      arch/arm64/kernel/vdso/vdso.so.dbg
  VDSOSYM include/generated/vdso-offsets.h

```

christianhaitian, 2024-08-02:
I found the issue and pushed the change to the git.  Please re clone or pull in the changes and try again.

imagemlt, 2024-08-02:
This time it works well. Thanks very much for this awesome project!

christianhaitian, 2024-08-02:
Awesome.  Enjoy

closed 2024-08-02 by christianhaitian