christianhaitian/arkos #1139: Bluetooth Gamepad bug on RGB30

hiruko-san, 2024-08-04:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Not related.

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Hotkeys and RGB30's input stops working in game if I pair with a bluetooth gamepad.

### How can the issue be reproduced? (If applicable)
After I paired the controller via bluetooth, I started a PS1 game (tested with Parasite Eve & Mega Man x4) running with Retroarch32. I couldn't use the hotkeys (e.g Select + X, or top face button) to go into RA menu. The inputs on the RGB30 will also stop working. I can't power off the device using the power button either. The only way is to press the Reset button to reset the whole device. This automatically unpairs the bluetooth gamepad.

I tested this with the XiaoMi bluetooth gamepad only so I'm not sure if other controllers are the same.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
07/31/2024

### Anything else you'd like to include that may help to review this issue or feature request?
I noticed the inputs configured via the Main Menu> Configure Input shows one set of values / numbers, but shows a different value if configured manually in Retroarch. I wasn't able to use the controller at all previously but seems like parts of the issue was fixed between the last version I tried and now the latest version. 

I poked around in the system folders and found 2 .cfg files referring to the same controller but one in chinese "小米蓝牙手柄" and one in English "Xiaomi Mi Bluetooth Gamepad".

christianhaitian, 2024-08-05:
Not sure how to even begin to troubleshoot this one as I don't experience that issue with the 8bitdo controllers I use.  It could be an issue with using Xiaomi controllers but I don't have any to test this.

hiruko-san, 2024-08-05:
I have the controller and when I last tested it I noted down the input values, if there's any help? 

[image]

to the left is the values from ES access from the Main Menu> Configure Input.
the blank spots under the Retroarch column on the right means the values are unchanged.

hiruko-san, 2024-08-16:
ok, so I got a 8Bitdo SN30 Pro, and I tested it again after I reset the input settings.
I've tried all 4 modes (DInput, XInput, Switch, Mac OS), the controller works normally while in the Emulation Station menu, but none of it worked perfectly in game. 

On PSX specifically I noted the Dpad, L1, and R1 weren't working.

On SNES & NES the Input just didn't work at all.

On Genesis only one of the ABXY face button responded in game.

christianhaitian, 2024-08-17:
Try updating the controller definitions in retroarch and see if that makes a difference.
<retroarch menu>online updater>update controller profiles>

hiruko-san, 2024-08-18:
> Try updating the controller definitions in retroarch and see if that makes a difference. online updater>update controller profiles>

unfortunately this didn't resolve it.

for PSX specifically the core remap file will load automatically each time and I have no idea how to disable the remap file.

apf3l, 2024-09-28:
I experience a similar problem with my RGB30 and bluetooth paired SNES Switch Online Controller or 8bitdo Pro 2 Controller L and R Button can be mapped in emustation frontend but when within SNES9x Emulator and Retroarch L and R is not working, already updated controller profiles and I am on ArkOS 09272024

apf3l, 2025-02-06:
I tried it again today using an Switch Pro Controller. I paired it using bluetooth and everything worked fine out of the box. All buttons worked in ArkOs and all buttons worked in retroarch as expected including L + R buttons.

Then I went to the controller input mapping in the main menu of ArkOs and remapped the inputs. In ArkOs everything still worked fine but in retroarch the L + R button stopped working. Which files are edited when the input mapping of ArkOs is done?