christianhaitian/arkos #1140: [Feature Request] Update the fbnneo core please

mzy1996, 2024-08-07:
Please update the Final Burn Neo core to fbneo latest commit [c9b7683](https://github.com/finalburnneo/FBNeo/commit/c9b768313edab44f61e0a7298a71275d4ab22441),it updates a lot of new roms support.

christianhaitian, 2024-08-07:
The core has been updated in the repo. Just update the core in retroarch via wifi. Thanks

mzy1996, 2024-08-07:
can I manual download the latest core from computer？

christianhaitian, 2024-08-07:
It will be a little bit of a process, but yes you can do that. 

Get it from here: https://github.com/christianhaitian/retroarch-cores/raw/master/aarch64/fbneo_libretro.so.zip

Take the fbneo_libretro.so file out of the zip and place it on your sd card.  Maybe the backups folder on the roms card will be best to place it in so it's easy to find.  

Boot the device, then use 351Files from the Options section to move the file to the `/home/ark/.config/retroarch/cores/` folder overwriting the existing one there.

mzy1996, 2024-08-07:
thanks，successful updated the core.

mzy1996, 2024-08-18:
Please update the Final Burn Neo core to fbneo latest commit [48c94b0](https://github.com/finalburnneo/FBNeo/commit/48c94b013b8772827045e0f19b7865e101973790),it updates a lot of new roms support.
my device is R35S