christianhaitian/arkos #1143: Not being able to download RG351MP GDrive

miltos19, 2024-08-13:
Hello,

When trying to download the file RG351MP GDrive i get the following message.

''Sorry, you can't view or download this file at this time.

Too many users have viewed or downloaded this file recently. Please try accessing the file again later. If the file you are trying to access is particularly large or is shared with many people, it may take up to 24 hours to be able to view or download the file. If you still can't access a file after 24 hours, contact your domain administrator.''

Is there something we can do to fix this issue?
Is there any other link that we can use?

Thank you for your time and consideration.

Kind regards,
Miltos

christianhaitian, 2024-08-13:
Did you try the mega link that is also available from the main wiki oage?

miltos19, 2024-08-13:
Yes but it was saying that it was a big file that I couldnt download directly but I needed the app in order to do it, so I didn’t proceed

christianhaitian, 2024-08-13:
You don't need to use the app to download.  Just do a standard download.  Other than that, the only other option is to wait for the google link to be available again.

closed 2024-08-13 by christianhaitian