christianhaitian/arkos #1145: [BUG] Ardens core not in latest version

sakamoto-neko, 2024-08-15:
Version: v2.0 (07/31/2024) Update
Device: RG353V

I've seen in the wiki that Arduboy games run on lr-ardens by default and lr-arduous as an alternate core. However I realized the lr-ardens core is not used in the latest update, so I was wondering if it was an oversight.
[image]
[image]
[image]

christianhaitian, 2024-08-15:
It's there.  Set the Emulator to Retroarch, then you can pick your core.

closed 2024-08-15 by christianhaitian

sakamoto-neko, 2024-08-15:
Ok my bad, this one's on me. I grabbed the GDrive link from the main wiki page and it had the 01/27/2024 update. I manually updated to the actual latest and it does use Ardens by default. Thanks for clarifying.