christianhaitian/arkos #1146: [Feature Request] Emulator for Epoch Super Cassette Vision

sakamoto-neko, 2024-08-16:
Firstly thanks for your work in ArkOS, it's my go-to firmware for my handheld devices now. I'd like to ask if this core could be implemented:

https://gitlab.com/MaaaX-EmuSCV/libretro-emuscv

The console does use a numpad, this core gets around it by using a button to open up a numpad menu. Other than that the controllers are 4 directions and 2 buttons only. Appreciate the response.

christianhaitian, 2024-08-16:
Looks interesting.  I'll build it and test it and if it seems to run without much issue, i'll include it in the next update.

Posting this youtube link to a review of this system here mostly for my reference.
https://www.youtube.com/watch?v=FP_yiC2iW5g

christianhaitian, 2024-08-17:
I was able to build the emulator and confirm functionality.  It will be included in the next update.  Thanks for the suggestion.

christianhaitian, 2024-08-24:
Added with yesterday's update.

closed 2024-08-24 by christianhaitian