christianhaitian/arkos #1147: Turn off LED

javinair, 2024-08-17:
Hi there.
Is it possible to turn the LED off completely (vía command/script/configuration)?
I'm using arkOS v2.0 on my R36S, I've seen it is possible to toggle red/blue colours (and some commands appears on screen).

I'm asking ir because playing at full darkness it's a little bit annoying.
[image]

Thanks.

christianhaitian, 2024-08-17:
I don’t support this device but I don’t think it’s possible to turn off that led since it’s a clone of the rg351mp and it it’s not possible there either.

closed 2024-08-17 by christianhaitian

bysonproductions, 2024-09-14:
Apologies if this is a dumb question - any chance we would get the ability to turn off the LED for the RGB20sx when in sleep mode? If that's not possible as well, understood, just thought I'd ask.

lauren7ino, 2024-09-14:
Suggestion: Once I fixed a similar problem in my living room (laser-like blue led) with drop of black nail polish. Keeps the light on, but much softer on the eyes.

bysonproductions, 2024-09-14:
100% doing something similar if I can't do it via software