christianhaitian/arkos #1148: Kodi not working

lauren7ino, 2024-08-17:
Hi
Kodi shows the spash screen and 1/2 second later closes back to ES.
Is there any error log I can send you?
Thanks

christianhaitian, 2024-08-17:
Try ssh into the unit, quit emulationstation then run kodi.sh manually.  See what errors it gives there.

lauren7ino, 2024-08-18:
ark@rg353m:/usr/local/bin$ Kodi.sh

Starting Kodi.  Please wait...fopen(): No such file or directory
/usr/local/bin/Kodi.sh: line 32: 19340 Segmentation fault      /opt/kodi/lib/kodi/kodi-gbm
ark@rg353m:/usr/local/bin$

lauren7ino, 2024-08-18:
With sudo

ark@rg353m:/usr/local/bin$ sudo Kodi.sh

Starting Kodi.  Please wait...fopen(): No such file or directory
libva info: VA-API version 1.5.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
/usr/local/bin/Kodi.sh: line 32: 20332 Segmentation fault      /opt/kodi/lib/kodi/kodi-gbm
/usr/local/bin/Kodi.sh: line 48: 20331 Killed                  sudo /opt/quitter/oga_controls kodi-gbm rg503

lauren7ino, 2024-08-18:
Verbose mode

ark@rg353m:/usr/local/bin$ sh -x Kodi.sh
+ export KODI_HOME=/opt/kodi/share/kodi
+ export PYTHONPATH=/lib/python3.7/
+ sudo perfmax kodi
+ export MALLOC_MMAP_THRESHOLD_=131072
+ [[ -e /dev/input/by-path/platform-fe5b0000.i2c-event ]]
Kodi.sh: 14: [[: not found
+ sudo chown root:root /dev/input/event2
+ sudo chown root:root /dev/input/event0
+ printf \n\n\e[32mStarting Kodi.  Please wait...

Starting Kodi.  Please wait...+ export LD_LIBRARY_PATH=/opt/kodi/libs/:/opt/kodi/libs/samba
+ cd /opt/kodi
+ + [[ !sudo -d /opt/quitter/oga_controls /home/ark/.kodi kodi-gbm ]] rg503

Kodi.sh: 27: [[: not found
+ /opt/kodi/lib/kodi/kodi-gbm
fopen(): No such file or directory
Segmentation fault
+ pidof oga_controls
+ [[ ! -z 24178 22616 ]]
Kodi.sh: 36: [[: not found
+ sudo perfnorm
+ printf \033c
+ sudo systemctl restart oga_events
ark@rg353m:/usr/local/bin$

christianhaitian, 2024-08-18:
Did you make changes to your setup recently?  Seems like the kodi folder is missing some content or some other change occurred.

lauren7ino, 2024-08-18:
Nothing I can remember. 
I confess I usually don't run Kodi, so I don't know if it's a new issue.
Many months ago I moved ArkOS to the internal disk, the run updates as usual.

I also see I'm not the only one in Discord...
[image]
[image]

christianhaitian, 2024-08-18:
Those issues were resolved when Kodi was updated to version 20.5.  It runs fine on all the rk3566 devices on my end on both emmc and sd card.  Did you apt install anything recently?

christianhaitian, 2024-08-18:
Did you make sure to update to the latest release via ota?

lauren7ino, 2024-08-18:
Yes, apt update and install Syncthing (on another thread). But I removed it, since it was not abre to overcome the connection errors.

christianhaitian, 2024-08-18:
That may have caused the issue.  Maybe it changed something that Kodi needs to run.

lauren7ino, 2024-08-18:
Is there an apt to reinstall kodi ?

lauren7ino, 2024-08-18:
All up to date. It says "No more updates available. Check back later."

christianhaitian, 2024-08-18:
Not for the Kodi used in this setup.  What process did you follow for installing syncthing?

lauren7ino, 2024-08-18:
From https://lenord.me/posts/syncthing_arkos/

sudo curl -o /usr/share/keyrings/syncthing-archive-keyring.gpg https://syncthing.net/release-key.gpg
echo "deb [signed-by=/usr/share/keyrings/syncthing-archive-keyring.gpg] https://apt.syncthing.net/ syncthing stable" | sudo tee /etc/apt/sources.list.d/syncthing.list
sudo apt update
sudo apt install syncthing

christianhaitian, 2024-08-18:
I just tried those steps and Kodi still works on my end so that doesn't seem to be issue.

lauren7ino, 2024-08-18:
Oh well, It was worth the effort 😊
Maybe one day someone figures it out.
Thanks, keep up the good work.

christianhaitian, 2024-08-18:
Problem is, it works for most people including myself.  I get quite a bit of feedback that it works fine for those that try it.

You can try the following via ssh and see what you get back

```
export KODI_HOME=/opt/kodi/share/kodi
export PYTHONPATH=/lib/python3.7/
export LD_LIBRARY_PATH=/opt/kodi/libs/:/opt/kodi/libs/samba

/opt/kodi/lib/kodi/kodi-gbm
```

lauren7ino, 2024-08-18:
ark@rg353m:/$ Connection to 192.168.1.147 closed by remote host.
Connection to 192.168.1.147 closed.
lauren7ino@Mac-mini ~ % ssh ark@192.168.1.147
ark@192.168.1.147's password: 

Welcome to ArkOS 2.0 (07312024)
Based on Ubuntu 19.10 (GNU/Linux 4.19.172 aarch64)

 * ArkOS Documentation:          https://github.com/christianhaitian/arkos/wiki
 * Ubuntu 19.10 Docs:            https://wiki.ubuntu.com/EoanErmine/ReleaseNotes
 * Ubuntu Documentation:         https://help.ubuntu.com
 * Ubuntu Management:            https://landscape.canonical.com
 * Ubuntu Support:               https://ubuntu.com/advantage

Last login: Sun Aug 18 15:11:27 2024 from 192.168.1.112
ark@rg353m:~$ cd /
ark@rg353m:/$ export KODI_HOME=/opt/kodi/share/kodi
ark@rg353m:/$ export PYTHONPATH=/lib/python3.7/
ark@rg353m:/$ export LD_LIBRARY_PATH=/opt/kodi/libs/:/opt/kodi/libs/samba
ark@rg353m:/$ /opt/kodi/lib/kodi/kodi-gbm
Segmentation fault

lauren7ino, 2024-08-18:
2nd try

ark@rg353m:/$ sudo /opt/kodi/lib/kodi/kodi-gbm
/opt/kodi/lib/kodi/kodi-gbm: error while loading shared libraries: libfstrcmp.so.0: cannot open shared object file: No such file or directory

lauren7ino, 2024-08-18:
Similar problem here: https://discourse.osmc.tv/t/kodi-leia-installation-error/73831

christianhaitian, 2024-08-18:
sudo won't work for launching Kodi on here.  That segmentation fault is odd.  Do the following:
```
sudo apt install strace
strace /opt/kodi/lib/kodi/kodi-gbm
```

christianhaitian, 2024-08-18:
Let me know what output you get from the strace

lauren7ino, 2024-08-18:
lauren7ino@Mac-mini ~ % sudo apt install strace
The operation couldn’t be completed. Unable to locate a Java Runtime.
Please visit http://www.java.com for information on installing Java.

christianhaitian, 2024-08-18:
Do this via ssh on the device, not your mac

lauren7ino, 2024-08-18:
lol! Sorry, the heat it frying my brain...

lauren7ino, 2024-08-18:
lauren7ino@Mac-mini ~ % ssh ark@192.168.1.147  
ark@192.168.1.147's password: 

Welcome to ArkOS 2.0 (07312024)
Based on Ubuntu 19.10 (GNU/Linux 4.19.172 aarch64)

 * ArkOS Documentation:          https://github.com/christianhaitian/arkos/wiki
 * Ubuntu 19.10 Docs:            https://wiki.ubuntu.com/EoanErmine/ReleaseNotes
 * Ubuntu Documentation:         https://help.ubuntu.com
 * Ubuntu Management:            https://landscape.canonical.com
 * Ubuntu Support:               https://ubuntu.com/advantage

Last login: Sun Aug 18 15:21:31 2024 from 192.168.1.112
ark@rg353m:~$ sudo apt install strace
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  strace
0 upgraded, 1 newly installed, 0 to remove and 6 not upgraded.
Need to get 275 kB of archives.
After this operation, 1,267 kB of additional disk space will be used.
Get:1 http://old-releases.ubuntu.com/ubuntu eoan/main arm64 strace arm64 4.26-0.2ubuntu3 [275 kB]
Fetched 275 kB in 3s (84.8 kB/s)
Selecting previously unselected package strace.
(Reading database ... 86451 files and directories currently installed.)
Preparing to unpack .../strace_4.26-0.2ubuntu3_arm64.deb ...
Unpacking strace (4.26-0.2ubuntu3) ...
Setting up strace (4.26-0.2ubuntu3) ...
ark@rg353m:~$ strace /opt/kodi/lib/kodi/kodi-gbm
execve("/opt/kodi/lib/kodi/kodi-gbm", ["/opt/kodi/lib/kodi/kodi-gbm"], 0x7fdf6a67e0 /* 23 vars */) = 0
brk(NULL)                               = 0x15da6000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=84398, ...}) = 0
mmap(NULL, 84398, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7facfc7000
close(3)                                = 0
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/aarch64/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/aarch64/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/aarch64/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/aarch64", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/tls", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/aarch64/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/aarch64/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/aarch64/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/aarch64", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/lib/aarch64-linux-gnu", {st_mode=S_IFDIR|0755, st_size=61440, ...}, 0) = 0
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/aarch64/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/aarch64/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/aarch64/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/aarch64", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/tls", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/aarch64/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/aarch64/atomics", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/aarch64/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/aarch64", 0x7ffdd88450, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/atomics/libfstrcmp.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
writev(2, [{iov_base="/opt/kodi/lib/kodi/kodi-gbm", iov_len=27}, {iov_base=": ", iov_len=2}, {iov_base="error while loading shared libra"..., iov_len=36}, {iov_base=": ", iov_len=2}, {iov_base="libfstrcmp.so.0", iov_len=15}, {iov_base=": ", iov_len=2}, {iov_base="cannot open shared object file", iov_len=30}, {iov_base=": ", iov_len=2}, {iov_base="No such file or directory", iov_len=25}, {iov_base="\n", iov_len=1}], 10/opt/kodi/lib/kodi/kodi-gbm: error while loading shared libraries: libfstrcmp.so.0: cannot open shared object file: No such file or directory
) = 142
exit_group(127)                         = ?
+++ exited with 127 +++
ark@rg353m:~$

christianhaitian, 2024-08-18:
Do the following via ssh and let me know the results:

```
export KODI_HOME=/opt/kodi/share/kodi
export PYTHONPATH=/lib/python3.7/
export LD_LIBRARY_PATH=/opt/kodi/libs/:/opt/kodi/libs/samba

strace /opt/kodi/lib/kodi/kodi-gbm
```

lauren7ino, 2024-08-18:
[output.txt](https://github.com/user-attachments/files/16649287/output.txt)

christianhaitian, 2024-08-18:
It seems like some of your Kodi files are missing.  Did you ever boot Kodi on your device successfully?

lauren7ino, 2024-08-18:
A long time ago. This year it may be the first time, not sure.

lauren7ino, 2024-08-18:
Can I extract all the files them from a repository or some zip file?

christianhaitian, 2024-08-18:
Yes.  Download all 3 files here https://github.com/christianhaitian/arkos/tree/main/03292024
Use 7zip to open the .zip file.  Copy the contents of /opt/kodi to your device’s install in the same location overwriting the existing files there. Might be easier to just create a backup of your existing settings and do a fresh install and update though especially if you’ve been doing quite a bit of tinkering.

lauren7ino, 2024-08-18:
I think messed up. I deleted opt/kodi/lib and opt/kodi/share previous to copying the new directories. My guess it that now some files are missing. Not even the splash screen shows anymore.
Better give up for today. Tired... Thanks for the support.

christianhaitian, 2024-08-18:
yea and plan to just backup your settings and restore to a new updated install.  You were not supposed to delete any existing files, just overwrite per the instructions I provided.  Good luck.

closed 2024-08-18 by christianhaitian

lauren7ino, 2024-08-18:
Finally restored the IMG Kodi and updated with the new files.
Still honest work, same error: error while loading shared libraries: libfstrcmp.so.0: cannot open shared object file: No such file or directory

I had to manually chmod +x Kodi. I wonder if there is something else related to permissions or symlinks missing (etc)

christianhaitian, 2024-08-19:
Just backup your settings and do a fresh install at this point is my recommendation.  Otherwise, you can try copying the /opt/kodi/libs content from the latest available image using 7zip.  Nothing else I can recommend on this.

lauren7ino, 2024-08-19:
OK thank you for the recommendation. I may well do that.
What is the correct procedure for a reinstall from the internal SSD to the same internal SSD?
Or do I need to boot ArkOS on an SDCard first? (can't find mine)
Thanks

christianhaitian, 2024-08-19:
1. Backup your current settings.
2. Follow the instructions here:https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353M#q-can-i-load-arkos-onto-the-internal-emmc-storage
3. Then do an OTA update from the emmc.
4. Finally, restore your settings.

lauren7ino, 2024-08-19:
It worked! Thanks!
Kodi is finally launching.

lauren7ino, 2024-08-20:
I must be the unluckiest guy in the world - not it's PortMaster that won't run.

ark@rg353m:/roms/tools$ ./portmaster.sh
p-cores: 0,1,2,3 (@1992.0000mhz)
cmd: taskset -c 0,1,2,3 
- extracting new pylibs.
Traceback (most recent call last):
  File "./pugwash", line 53, in <module>
    with zipfile.ZipFile(PYLIB_ZIP, 'r') as zf:
  File "/usr/lib/python3.7/zipfile.py", line 1225, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1292, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file