christianhaitian/arkos #1149: Emergency boot

lauren7ino, 2024-08-17:
Hi
I booted into the emergency boot mode because i selected an ES theme that won't load.
Restore backup is apparently not working.
351file shows backups both in internal and SD card.
I had to use 351Files do delete the ES config file so I could boot.

christianhaitian, 2024-08-17:
What message did you get when you attempted to do a restore backup?

lauren7ino, 2024-08-18:
None. The menu selection seems not to do anything.
I'm using ArkOS in the internal SSD.

lauren7ino, 2024-08-18:
Just made a new test. Deleted all backups (ROMS+ROMS2), and made a new backup.
Confirmed that the new gz files appear in both ROMS+ROMS2.
Rebooted with BaRT.
Neither option 6 not 7 work. Flashes 1/10th of second black then reloads BaRT.

christianhaitian, 2024-08-18:
Oh you’re right.  Those 2 options are not functioning.

christianhaitian, 2024-08-18:
I found the issue and will apply the fix with the next update.  You can fix this on your end by copying and pasting the following command via ssh then press enter.  Then reboot and confirm functionality.

```
[[ -z "$(cat /usr/bin/emulationstation/emulationstation.sh | grep '/opt/system/Advanced/"Backup ArkOS Settings.sh' | tr -d '\0')" ]] && sudo sed -i "/\"8)\") sudo reboot/s//\"6)\") sudo kill -9 \$(pidof boot_controls)\n                                \/opt\/system\/Advanced\/\"Backup ArkOS Settings.sh\" 2>\&1 > \/dev\/tty1\n                                sudo .\/boot_controls none \$param_device \&\n                                ;;\n                          \"7)\") sudo kill -9 \$(pidof boot_controls)\n                                \/opt\/system\/Advanced\/\"Restore ArkOS Settings.sh\" 2>\&1 > \/dev\/tty1\n                                sudo .\/boot_controls none \$param_device \&\n                                ;;\n                          \"8)\") sudo reboot/" /usr/bin/emulationstation/emulationstation.sh /usr/bin/emulationstation/emulationstation.sh.ra /usr/bin/emulationstation/emulationstation.sh.es
```

lauren7ino, 2024-08-18:
OK it started working. Good work!
Now, why does the restore settings delete the /roms/backup folder files?
What's the logic? Why can only restore once?

christianhaitian, 2024-08-18:
You don't have to delete the backup files.  You're given the option to delete it after a successful restore.  Perhaps you don't want the backup to remain there since it contains sensitive information like your wifi password.

lauren7ino, 2024-08-18:
OK. Sorry. My fault for not reading the full text.
Maybe the default should not be to delete, I don't know.

christianhaitian, 2024-08-18:
There is no default.  You press A to delete or B to not delete.  The choice is on the user.

christianhaitian, 2024-08-18:
If there was a choice, the choice would be security first, delete the file with the sensitive information.

christianhaitian, 2024-08-24:
Addressed with yesterday's update.

closed 2024-08-24 by christianhaitian