christianhaitian/arkos #1152: Doom Scanner doesn't respond to controls

VGkav, 2024-08-18:
Hello, I selected gzdoom-standalone as the Doom player and I threw doom2.wad inside the Doom folder, I had a "doom2" game and I launched it succesfully!

My issue is that I want to load pwads and the Doom Scanner script thingy loads but it doesn't respond to any keys, only start+select to exit. I am running the latest version on an rg503!

What is the proper way to launch pwads? Can I create .doom extension text files inside the folder and set them up according to the info I found here? Will that work?

https://github.com/plaidman/rgb10max/wiki/doom-loader

christianhaitian, 2024-08-18:
What version of ArkOS are you on?  You can see this at the bottom of the start menu next to Distro Version.

christianhaitian, 2024-08-18:
I just tested playing the batman mod from here with gzdoom on the latest version of arkos (07312024) and it worked just fine on my end.  I used the left joystick (not the dpad) to navigate the menu and start the game.  The dpad doesn't work with the menu.

christianhaitian, 2024-08-18:
My .doom file contained the following:

```
IWAD=DOOM2.WAD
MOD=mods/batman/batman.wad
DEH=mods/batman/batman.deh
SAVE=mods/batman
-- end --
```

VGkav, 2024-08-19:
I did an update just before messing with Doom, so it's the latest (yesterday). If it wasn't clear, Gzdoom works fine for me! I will create doom files by hand, good to know.

But my acthal issue is that the utility/script called doom-scanner (I think) is unusable, it doesn't respomd to controls! Shouldn't it work? What's its purpose? It seems to "create config"?

christianhaitian, 2024-08-19:
Doom-Scanner?  I don't even know what that is.  ArkOS doesn't include such a utility.

VGkav, 2024-08-19:
Weird! Inside roms2/doom I found a shell script with the name "_Scan Doom Games.sh"

And it has a few lines of commands which basically run doomscan. Is this an EmulationStation thing maybe?

closed 2024-08-19 by christianhaitian