christianhaitian/arkos #1153: ThemeMaster Preview

lauren7ino, 2024-08-18:
Hi,
Does ThemeMaster preview work in ArkOS?
https://github.com/JohnIrvine1433/ThemeMaster/issues/36#issuecomment-2295376933

christianhaitian, 2024-08-19:
It should but image-viewer is not displaying anything for some reason.  Shows no error upon execution and strace shows a bunch of errors related to files it's looking for.  I tried to compile my own version but it's resulting in the same issue.  replacing image-viewer with ffplay works fine though but not sure john would want to incorporate that.

JohnIrvine1433, 2024-08-19:
@christianhaitian the binary works just fine on latest ArkOS release for 351P, any idea what could have changed in latest ArkOS releases with SDL2 libraries?
PS : I've rewritten the image viewer as I had some problems running it on ROCKNIX, I've created a new repository for the new code : https://github.com/JohnIrvine1433/ThemeMaster-Image-Viewer

christianhaitian, 2024-08-20:
@JohnIrvine1433 Do you mean the latest forked release by Wummle or Slayer366 for the 351P?  I haven't updated ArkOS for the 351P since 2021.  For supported devices, SDL is currently at 2.30.3.  All I get is a black screen on there on supported rk3566 and rk3326 devices using the included image viewer or building from the original source.  I tried going back to previous versions SDL2 but same results with all of the ones I tried. 2.10, 2.16, 2.18.2 and so on.  Not sure what changed to cause this issue honestly.

JohnIrvine1433, 2024-08-23:
@christianhaitian I am still using your final version of ArkOS for 351P which has no issue with the image viewer. 
Does the fork from Wummle aligns with the content of latest releases ?

christianhaitian, 2024-08-23:
@JohnIrvine1433 I think so but i'm not 100 percent sure.

christianhaitian, 2024-10-07:
Any update on this issue?

JohnIrvine1433, 2024-10-10:
Well I have been able to rebuild a viewer that works just fine on ArkOS but then I have an issue with ROCKNIX...
Weston is really painful in terms of of window and focus management (well at least I can't figure it out on my end). 
I'll give it a last try this weekend, I might just end up pushing two different viewers. 
In short, should be fixed this weekend.

lauren7ino, 2024-10-10:
Maybe with a few #ifdef <ROCKNIX> ... #else ... #endif

JohnIrvine1433, 2024-10-11:
V5.0.0 just pushed as a pre-release.

lauren7ino, 2024-10-11:
Wow! A small feature that makes such a large difference!
I no longer have to install hundreds of themes to see them!
Thank you! 🙏🏼

christianhaitian, 2024-10-12:
Thanks John

closed 2024-10-12 by christianhaitian

lauren7ino, 2024-10-13:
Hi
I can't make the preview work on the rg353m, although works fin on the rgb30.
can you confirm and report back?

christianhaitian, 2024-10-16:
@JohnIrvine1433 The new legacy image-viewer doesn't seem to work on the 640x480 devices in ArkOS.  Here's a version that does.  It was built from this [source](https://github.com/christianhaitian/Image_Viewer).  This version does not have the ability to flip the image and I didn't bother to add that because I couldn't tell the value of adding such a feature for previewing themes.

https://github.com/user-attachments/files/17388802/image-viewer-legacy.zip