christianhaitian/arkos #1154: Improvement: dtb file

lauren7ino, 2024-08-19:
Improvement suggestion:
Backup and Restore the DTB file on use.
Specially useful for those people that swap the DTB file because of screen version.

christianhaitian, 2024-08-19:
I don't understand the request.

gabrielmop, 2024-08-20:
You can backup the DTB manually by connecting the SD card on your pc and copying the DTB files to your PC

lauren7ino, 2024-08-20:
Thanks, I know that. I'm using the internal eMMC, so it's not that linear.
The suggestion was to include the boot DTB file into the backup/restore scripts.

christianhaitian, 2024-08-21:
Not something i will incorporate at this time since the issue only affects a small portion of 353 devices with v1 screens.  I have a v1 353v that is on 24/7 for development and testing and v1 353m i use daily and they don’t exhibit this issue.

closed 2024-08-21 by christianhaitian