christianhaitian/arkos #1155: [Feature Request] J2ME emulator

gabrielmop, 2024-08-20:
It is possible to bring J2ME emulation to ArkOS?

christianhaitian, 2024-08-20:
grumble grumble grumble, i'm not a fan of jave games.  I felt they were all terrible.  I may look at this but not sure honestly.  It doesn't motivate me.

michidaeron, 2024-11-10:
> grumble grumble grumble, i'm not a fan of java games. I felt they were all terrible. I may look at this but not sure honestly. It doesn't motivate me.
I feel that it is the only thing missing to be complete when it comes to emulators. Games like Asphalt, CSI, Rock Band, Silent Hill, Metal Slug, Treasure Towers and many more were the best moments :C 
I installed j2me following a tutorial but some games don't work for me. I don't know if it would be the same in terms of a more complete implementation like yours. Anyway, thanks for your work.

christianhaitian, 2024-12-15:
I finally got interested enough to check this out.  Doom RPG works pretty well on it and is not bad actually. Quite surprising to me.  I haven’t tried any of the games listed above yet but will try a few of them and see what the results are.

christianhaitian, 2024-12-24:
This has been added with today's update.

closed 2024-12-24 by christianhaitian

raay8456, 2025-01-30:
Hello sir , I just updated to latest arkos and have put .jar files in the j2me folder , but the j2me don't show up on es-ds menu !! I can run individual game through retroarch  . Please give directions on how to play java games !!!!

raay8456, 2025-01-30:
Please give solution