christianhaitian/arkos #1157: PortMaster dones't launch

lauren7ino, 2024-08-21:
I already tested this procedure twice:

Install the latest IMG file into an SDCard, boot and OTA update.
PortMaster doesn't launch. It appears some files are missing.

ark@rg353m:/roms/tools$ ./portmaster.sh
p-cores: 0,1,2,3 (@1992.0000mhz)
cmd: taskset -c 0,1,2,3

extracting new pylibs.
Traceback (most recent call last):
File "./pugwash", line 53, in
with zipfile.ZipFile(PYLIB_ZIP, 'r') as zf:
File "/usr/lib/python3.7/zipfile.py", line 1225, in init
self._RealGetContents()
File "/usr/lib/python3.7/zipfile.py", line 1292, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

Original post: https://github.com/christianhaitian/arkos/issues/1148#issuecomment-2299697056

christianhaitian, 2024-08-21:
Are you using a separate sd card for your roms?

lauren7ino, 2024-08-21:
Yes—Laurentino MartinsEscrito no telemóvel, peço desculpa por algum erro.

christianhaitian, 2024-08-21:
If you're using a separate sd card for your roms, why is portmaster running from `/roms/tools` and not `/roms2/tools`?

lauren7ino, 2024-08-21:
In this test I probably didn’t activate the ROMS2 SDCard.
I will check later when I get back home.

lauren7ino, 2024-08-21:
You were right, it was the PortMaster on ROMS2 that was broken.
I replaced it with the PortMaster from ROMS and now it works again.
You can close this issue now.
I still have no idea what was the original cause of the problem though.

closed 2024-08-21 by christianhaitian