christianhaitian/arkos #1163: FAQ Edit

everynone-bit, 2024-08-27:
The FAQ includes an entry that I believe to be out of date. #30 relates to removing duplicates in the PS1 section and refers to an entry **PS1 - Show only m3u files.** that no longer exists inside of the ArkOS options. I believe it should be changed to:

1. Be sure to follow the steps [here for generating .m3u files in ArkOS](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353m#q-how-do-i-generate-m3u-files-for-ps1-games-in-arkos)
2. Navigate to the PS1 Section
3. Press **Start** scroll to **UI Settings**
4. **Theme Configuration** -> **File Extensions** 
5. only check **.m3u**

I'm not sure this is the right way to suggest an edit for the FAQ, but figured I'd save the next person the hour or two I spent searching for the option.

christianhaitian, 2024-08-27:
Ah yes.  That is correct.  Thanks for the catch.

christianhaitian, 2024-08-27:
Done.  Thank you again.