christianhaitian/arkos #1164: A simple question

lauren7ino, 2024-08-29:
Hi
I have an rg353m, and a rgb30 on my way from AliExpress.
For the new rgb30, I will like to configure it from scratch.

Q:
While it doesn't arrive, can I use the rg353m to configure the memory card(s) in advance?
I'm planning in using the rgb30 image on the rg353m, but swapping the dtb file, so that I can see what I'm doing.
Later, I will retore the original rgb30 dtb.

I know it boots. But is it safe?
Thanks

christianhaitian, 2024-08-29:
This may work.  I've not tried this though.  Worst case, you can probably do a settings backup and restore on a new setup if it doesn't workout.  Good luck.

closed 2024-08-29 by christianhaitian

lauren7ino, 2024-09-03:
Just to let you know that yes, it works.. Confirmed now that my RGB30 arrived. Thanks