christianhaitian/arkos #1168: Backup restore

lauren7ino, 2024-09-02:
Hi,
I installed ArkOS in a brand new SDCard and restored the backup from the old.
It seems to me some setting are shown to be set in Theme Master, but they are not.
I remember I had a couple of examples, but I forgot to write them down.
One I'm certain is "Switch button tap to off" that after restore it is still on "tap to sleep".
Another detail that should be improved, is a restore do a reboot/reload ES in the end.
Thanks, keep up the good work!

By the way, I'm sharing my benchmarks.
Not all SDCard are born equal. ArkOS boot with same backup restored:
  Original Kioxia 16gb: 23 seconds
  New Sandisk Extreme 64gb: 16 seconds

christianhaitian, 2024-09-02:
I'll need the specifics on which settings did not restore.  I won't force a reboot after the restore because depending on what the unit is used to restore, a reboot is not always necessary like in the case when I use it to have folks test a functionality that doesn't require the OS to reset.

lauren7ino, 2024-09-03:
The test it's easy to do. On the default ES settings, backup using the script, change every setting in ES to something else/opposite, and restore the backup.
Just tested it, 
"Switch button tap to off" says it's on, but goes to sleep,
"Game Loading image" says None, but it displays the default image,
"Verbal Battery Warning" says OFF, but still speaks the warnings,
etc.

As a developer, I believe every setting that (behind the scenes) runs a script on  toggle,
  1) The script must be run manually after the restore.
or
  2) The setting should not be restored.
Otherwise happens what is happening to me: I don't trust these settings anymore.
For ex "Search for Local Art" is set to On, but is it though???

Thanks

lauren7ino, 2024-09-03:
Also the time zone is not restored.

christianhaitian, 2024-09-03:
I've accounted for these missing restore options and will provide updated Backup and Restore tools in the next update.  Note that for rk3566 devices, I will not restore any set undervolt options due to the fact that not all devices are created equal when it comes to stability of setting those undervolts.

christianhaitian, 2024-09-28:
This has been addressed with today's update.

closed 2024-09-28 by christianhaitian