christianhaitian/arkos #1170: [Feature Request] Adding a few consoles and handhelds from lr-mess

sakamoto-neko, 2024-09-04:
I was wondering if it would be possible to add these consoles with lr-mess via Retroarch. I'm mainly testing the consoles on the Android side using a RG353V device. I did put some results below.

**TimeTop GameKing** - lr-mess Android OK
**TimeTop GameKing 3** - lr-mess Android NG, works on standalone MAME4droid 2024 (0.268)
**Hartung Game Master** - lr-mess Android OK
**Epoch Game Pocket Computer** - lr-mess Android OK
**Casio PV-1000** - lr-mess Android NG, does work on AmberElec
**Bit Corp Gamate** - lr-mess Android OK
**Casio PV-2000** - lr-mess Android OK, needs special controls

For Casio PV-2000 specifically, it emulates a keyboard on MAME/MESS, but I've figured out the controls for both players:

**PLAYER 1**
Enter > START
Yen > SELECT
Attack 0 > ATTACK 0
Attack 1 > ATTACK 1
Cursor Up > UP
Cursor Down > DOWN
Cursor Left > LEFT
Cursor Right > RIGHT

**PLAYER 2**
IN4_1 > START
IN4_2 > SELECT
IN8_2 > ATTACK 0
IN8_3 > ATTACK 1
IN7_3 > UP
IN6_2 > DOWN
IN7_2 > LEFT
IN6_3 > RIGHT

It's quite a big request, but thanks for taking the time to read it.

christianhaitian, 2024-09-04:
Have you tested any of these under ArkOS and confirmed functionality?

sakamoto-neko, 2024-09-04:
Is there an easy way to test the roms under ArkOS? I could test them and provide results.

christianhaitian, 2024-09-04:
Run them through retroarch directly.  There should be a retroarch and retroarch32 option under the retroarch system menu option.

sakamoto-neko, 2024-09-04:
Tested the systems via lr-mess on ArkOS:
**TimeTop GameKing** - OK
**TimeTop GameKing 3** - NG
**Hartung Game Master** - OK
**Epoch Game Pocket Computer** - crackly audio
**Casio PV-1000** - screen resolution incorrect, otherwise sound and controls OK
**Bit Corp Gamate** - OK
**Casio PV-2000** - have to set up controls but otherwise OK