christianhaitian/arkos #1173: Parse GamesLists Only

lauren7ino, 2024-09-09:
When we change this setting "Parse GamesLists Only", ES must restart.
Also maybe a bug, why there are so many tools missing when "Parse GamesLists Only" is set?

Reason:
I'm using "Parse GamesLists Only" because I have a hand curated gameslist.xml for Commodore Amiga.
My repository has both .m3u files for multi disk games, and .adf files for single disc games.
.adf files referred in the .m3u are not to be shown.
Thanks

christianhaitian, 2024-09-09:
This is expected behavior.  ES must restart in order to parse the existing list and not include any new data.  Options are parsed on boot but ignored from inclusion in GameLists so that they don't end up as part of the Last Played Games list.

closed 2024-09-09 by christianhaitian

christianhaitian, 2024-09-10:
Options are parsed on boot but ignored from inclusion in GameLists so that they don't end up as part of the Last Played Games list.

lauren7ino, 2024-09-14:
Hi
Have you seen the screenshots above?
Parse gamelists only hides many Options after a reboot.
Is this intended?